ReneLergner/WPinternals #4: Request: Support for OS version 10.0.15254.562

JamePeng, 2019-04-17:
Microsoft was push the new W10M system 10.0.15254.562 this month, but the WPinternals notice said that was not support this version.

Aarklendoia, 2019-05-22:
The lastest version is now 10.0.15254.566 !

AxelPanda, 2019-06-12:
unfortunately, the latest version is 10.0.15254.572 now...
is there any way to prevent windows from updating?

trochdewei, 2019-06-30:
> is there any way to prevent windows from updating?

there must be registry key. I don't remember exactly which. You can edit registry using interop tools

davepermen, 2019-07-16:
10.0.15254.575 now..

Aarklendoia, 2019-07-16:
For information, the last version of WPInternals supports all updates of Windows (current and future). It's only a problem to enable the root access, not to unlock the bootloader.

davepermen, 2019-07-19:
Root access is the one needed to extract the WhatsApp messages.db to move it over to Android, so I guess that's one reason there's interest specifically towards the root access.

mandrearg, 2019-08-02:
I'm interested too in support for this Windows Mobile 10 build, I wasn't able to root access my Lumia 550 becasue this build isn't supported by WPInternals 2.8

vluizflavio, 2019-08-09:
Also does not support Lumia 640XL, OS version 10.0.15063.1868

gersteba, 2019-12-01:
Please support newest version 10.0.15254.597, too - thank you!

quiret, 2019-12-09:
Please read my discussion with Csybormelgen at woaproject.net to fix problems due to outdated wpinternals (because it lacks support for latest W10M builds):

https://www.woaproject.net/viewtopic.php?f=17&t=85#p489

MagicAndre1981, 2019-12-13:
The final build of Windows 10 mobile is 10.0.15254.600. 

any chance to get an up to date version @gus33000 @ReneLergner

SIRprise, 2019-12-27:
I don't remember exactly what I did (and my phone is now bricked), but to get the WhatsApp database and other stuff I did not need the root access:
- unlocked the bootloader and backuped everything
- saved bin-files of the filesystems
- enabled mass storage mode with access to filesystem, changed permissions to access the whatsapp folder and copied it. 

I think the last step was not neccessary: With the .bin-file you can use the tool IsoBuster and browse through the filesystem.
A german version of what I did is here:
[https://www.wpvision.de/ich-brauche-hilfe-bei/11347-lumia-650-dateisystem-zugriff.html](url)

gus33000, 2019-12-30:
Root Access isn't required to access files on the phone. You can do that using Mass Storage.

hanlong85, 2020-01-15:
> I don't remember exactly what I did (and my phone is now bricked), but to get the WhatsApp database and other stuff I did not need the root access:
>
> * unlocked the bootloader and backuped everything [...]

where to find the bin-files?

tkjune, 2020-04-05:
> I hate to be that guy, but on my phone I read version **10.0.15254.603**.
> As far as release notes tell me, I suppose version 2.8 of WPInternals is not compatible with this WP build, right?

Tested... 10.0.15254.603 not supported :(

AMBNETO, 2020-04-12:
It maybe not 100% supported, but I managed to unlock my bootloader using WPInternals 2.8 (version 15254.603). It will not download the emergency files, you will need to download generic ones over the internet. But that it's, then you click on Unlock Bootloader and it will work. If the process fails, try again. I tried 4 times and it worked.

gus33000, 2021-02-08:
Support for the latest build of Windows phone was added by a24998c #15 
I am closing this issue as a result, if you feel like support for older revisions need to be added, feel free to reply on this thread and I might reopen it.

closed 2021-02-08 by gus33000