ReneLergner/WPinternals #5: Error: Failed to read sector. Exception: 0x00000002

Hemp007, 2019-05-19:
Не разблокируется загрузчик. В чем может быть проблема?
[image]
[image]

gus33000, 2019-12-30:
Can you please provide more information in english if possible?

gus33000, 2021-08-11:
Hi,

We didn't hear back from you in ages, are you still having this issue? We might have to close this issue if we don't hear back from you in a few days...

(New version: https://github.com/ReneLergner/WPinternals/releases/tag/2.9)

garhold, 2022-05-28:
The same issue here with Nokia Lumia RM-1118 in WPinternals 2.9.2 win-x64.

It happens when rebooting the phone in Mass Storage mode.

Windows offers to format the disk which of course I do not.

In device manager 40 partitions appear related to the phone.

Event ID 11 is logged in Windows 'Driver detected a controller error'.

Is there a work around to this problem ?

[image]

[image]

closed 2024-10-12 by gus33000