ReneLergner/WPinternals #6: Ambiguous error message "Bad GPT"

quiret, 2019-12-09:
If you start the Unlock process of the Lumia 950 boot sector then you can receive a fatal error that says "BAD GPT". [Please take a look at this discussion over at woaproject.net.](https://www.woaproject.net/viewtopic.php?f=17&t=122).

It could be a good idea to distiguish between BAD GPT inside ffu and BAD GPT requested from the Lumia 950 smartphone, since both are likely error scenarios.

gus33000, 2019-12-30:
Will implement error catching inside the unlock process as the exception is thrown by the GPT class, and not anything else external to it. I already improved error messages with more details but this doesn't address this.

cantcodegood, 2020-01-24:
Hey @gus33000, I am also receiving this on an unlocked (and unbranded) 950 XL.  I tried deactivating device encryption with no change, then I tried a full device reset with the same result, then I used the Windows Device Recovery Tool to reinstall the OS version with the same result.

Files downloaded through wpinternals application:

- MPRG8994_fh.ede
- RM1116_1078.0053.10586.13169.15357.034F05_retail_prod_signed.ffu
- RM1116_fh.edp

[image]
[image]

gus33000, 2021-08-11:
Fixed in 2.9
https://github.com/ReneLergner/WPinternals/releases/tag/2.9

Converseallstar95, 2021-08-19:
> Fixed in 2.9
> https://github.com/ReneLergner/WPinternals/releases/tag/2.9

Hi, I'm having the same issue as cantcodegood with my RM-1116. Fresh reset, no encryption, WPInternals 2.9.2, bad GPT.

Also, when downloading all the files prior, after it downloads the 3 RM-1116 files, it downloads the .ffu for RM-1085. Is this by design?

Arlodotexe, 2021-08-19:
I'm seeing the same thing in 2.9.2. It downloads files for RM-1105 for my Lumia 950 and displays "BAD GPT" when trying using them to unlock the bootloader.

gus33000, 2021-08-19:
@Converseallstar95 @Arlodotexe I would try manually downloading the FFU file from https://lumiafirmware.com/ then, this is thrown whenever the FFU file is actually corrupt (the tool is unable to parse the GPT binary as a result), tell me if it works, reopening for now

gus33000, 2021-08-19:
> Also, when downloading all the files prior, after it downloads the 3 RM-1116 files, it downloads the .ffu for RM-1085. Is this by design?

Yes, if the FFU file is not compatible with the patches the tool needs to make for unlocking the bootloader, it will always download a fall back version to extract that specific file from and patch it

Converseallstar95, 2021-08-19:
> @Converseallstar95 @Arlodotexe I would try manually downloading the FFU file from https://lumiafirmware.com/ then, this is thrown whenever the FFU file is actually corrupt (the tool is unable to parse the GPT binary as a result), tell me if it works, reopening for now

Thanks for getting back so quick, I tried downloading from that site, yet every firmware I click says "Error: Not found!" 
I've tried multiple browsers with varying addons enabled.

gus33000, 2021-08-19:
> Thanks for getting back so quick, I tried downloading from that site, yet every firmware I click says "Error: Not found!"
> I've tried multiple browsers with varying addons enabled.

I see, let me attach an offline tool that will generate links for them then, sorry about that.

Help:

```

Software Repository (SoRe) Fetch Utility
Copyright 2021 (c) Gustave Monce

This software uses code from the following open source projects released under the MIT license:

WPinternals - Copyright (c) 2018, Rene Lergner - wpinternals.net - @Heathcliff74xda

Please see 3RDPARTY.txt for more information

SoReFetch 1.0.0
Copyright (C) 2021 SoReFetch

ERROR(S):
  Required option 't, product-type' is missing.

  -c, --product-code         Product Code.

  -t, --product-type         Required. Product Type.

  -o, --operator-code        Operator Code.

  -r, --revision             Revision (useful for getting older package versions).

  -f, --firmware-revision    Phone Firmware Revision (useful only for changing what Test package you get for
                             ENOSW/MMOS).

  --help                     Display this help screen.

  --version                  Display version information.

```

[SoReFetch.zip](https://github.com/ReneLergner/WPinternals/files/7017189/SoReFetch.zip)

Arlodotexe, 2021-08-19:
Thanks for the quick reply! Here's the error message returned by the supplied tool:

```
PS B:\bin\win\general\SoReFetch> .\SoReFetch.exe -c 059X4D5 -f 01078.00053.16236.35035 -o ATT-US -t Lumia 950

Software Repository (SoRe) Fetch Utility
Copyright 2021 (c) Gustave Monce

This software uses code from the following open source projects released under the MIT license:

WPinternals - Copyright (c) 2018, Rene Lergner - wpinternals.net - @Heathcliff74xda

Please see 3RDPARTY.txt for more information

Searching FFU

Revision found (FFU): 10095

Searching ENOSW

Error while searching ENOSW
ENOSW package not found
No ENOSW package has been found in the remote software repository for the requested model.

Searching Emergency

Getting Emergency files for: LUMIA
Error while searching Emergency
'doctype' is an unexpected token. The expected token is 'DOCTYPE'. Line 1, position 3.

Results

Product Type: RM-1105

FFU: https://softwarerepo.blob.core.windows.net/softwarerepo/52412085-6311-47fe-bc32-8133fda3b12d/RM1105_1078.0053.10586.13169.12716.035044_retail_prod_signed.ffu?sig=to6o%2BQipEFtcaVpSeTGgKAaDEXUr1MktiXPa%2Bs6tNFc%3D&se=2021-08-20T19%3A12%3A30Z&sv=2014-02-14&sp=r&sr=b
```

gus33000, 2021-08-19:
I see, the error you got is mainly because -t supposedly should have been RM-1105 in this case but the server will fallback with the product code either case, that's why you got a successful link in the end.

Converseallstar95, 2021-08-19:
Hm, mine didn't work, I typed "-c RM-1116 -t 059X505"

I got FFU not found, ENOSW not found

gus33000, 2021-08-19:
> Hm, mine didn't work, I typed "-c RM-1116 -t 059X505"
> 
> I got FFU not found, ENOSW not found

You swapped -t and -c, swap both values and try again.

Converseallstar95, 2021-08-19:
> > Hm, mine didn't work, I typed "-c RM-1116 -t 059X505"
> > I got FFU not found, ENOSW not found
> 
> You swapped -t and -c, swap both values and try again.

Ah, thanks for that. I got 4 links, I'll download them all and put them in the WP ProgramData folder/report back after trying again!

Arlodotexe, 2021-08-19:
> I see, the error you got is mainly because -t supposedly should have been RM-1105 in this case but the server will fallback with the product code either case, that's why you got a successful link in the end. I will probably open source the tool this weekend given not a lot of solutions exist, perhaps adding a way to get links from wpinternals might be a better idea.

Ah, that makes sense. If you'll be working on this this weekend -- a bit of docs to say this is the "manufacturerHardwareModel" could be helpful.

Converseallstar95, 2021-08-19:
Successfully unlocked the bootloader; thank you!

Arlodotexe, 2021-08-20:
It works, big thanks Gus!
[image]

gus33000, 2021-08-21:
Just pushed the repository for the tool here: https://github.com/gus33000/SoReFetch

Will do more documentation/work on it in the coming hours. I think this is also needed as a feature in WPinternals (right click, get a link, or something a bit more clear)

Killerhacker200, 2021-10-09:
> > Does this still open? I have a lumia 640 RM-1077 || 059X137 || 000-VN. I downloaded the FFU and ede in the download section of WP Internal, when i unlock bootloader, It says Bad GPT. When i use the FFU on lumiafirmware, it requires a donor ffu. Anyone got any ideas to fix this?
> > Edit: this is the downloaded version on lumiafirmware: RM1077_02177.00000.15184.36002_RETAIL_prod_signed_1006_0248BF_000-VN.ffu
> 
> You need to download a RM-1085 ffu and this will be your donor FFU (read text on the unlock page as to why and more instructions on the subject). Leave your RM1077 as normal FFU.

It still doesn't work. What did i do wrong?
[image]

gus33000, 2021-10-09:
You need to download RM-1085/hwid/059X4T0 01078.00053.16236.35018

Killerhacker200, 2021-10-10:
thanks!! 
[image]

Killerhacker200, 2021-10-10:
uhm, now it says bitlocker key ?? im stuck in a blue screen 0_0

Killerhacker200, 2021-10-10:
here we go, start from phase1
[image]

Killerhacker200, 2021-10-11:
> > uhm, now it says bitlocker key ?? im stuck in a blue screen 0_0
> 
> For future reference, you need to turn off device encryption before starting the unlock process.

thanks! I finally unlocked the bootloader.

Killerhacker200, 2021-10-11:
off topic: any ideas to input arrows, enter to lumia during a pre-OS stage?

oleggamer351, 2024-04-27:
I have a lumia 640 RM-1077 and i selected downloaded ffu and ede it says this
[image]
[image]

gus33000, 2024-04-28:
Your ffu file os corrupted or not downloaded fully. Please redownload your file and try again and it should work fine thereafter :)

oleggamer351, 2024-05-08:
and how to fix this with ffu profile search Failed to write to pipe: 4E4F4B58464645
[image]

oleggamer351, 2024-05-09:
and how to fix this 
[image]

Omarovsky, 2024-05-20:
> and how to fix this with ffu profile search Failed to write to pipe: 4E4F4B58464645 ![Screenshot_14](https://private-user-images.githubusercontent.com/131574761/329084302-a6e20c3e-62c7-410d-aece-8bcd2db05441.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYyMjQ1NzcsIm5iZiI6MTcxNjIyNDI3NywicGF0aCI6Ii8xMzE1NzQ3NjEvMzI5MDg0MzAyLWE2ZTIwYzNlLTYyYzctNDEwZC1hZWNlLThiY2QyZGIwNTQ0MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyMFQxNjU3NTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xOTZhM2M4ZGYzNzg5OTM4MTA5MzJiMWU5MzhiOWE1YmI5YzhiN2NjMDIyZWNiZGY1NTA5ZDgyZTY3OTgyZjYyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.slo9_HZO_8Nt1g_8hhf9aOcSROYBBobOJbpFvSCeXto)

i have same problem

gus33000, 2024-10-12:
FFU download functionality is deprecated.

closed 2024-10-12 by gus33000