ReneLergner/WPinternals #7: Version 2.3 works but Version 2.8 does not

Revelation1714, 2019-12-29:
I was having an issue with my Lumia 950 (059X4D5 RM-1105). Windows Phone Internals version 2.8 went through the normal steps of looking for a flash profile, but NEVER found it. And gave an error message at the end.
After several failed attempts, and trying to unlock bootloader, I decided to try different versions of WP Internals.
When I got to version 2.3, it worked flawlessly, and went straight into flash mode.
Before I did anything else, I tried 2.8 again to make sure it was not a coincident.
It had the same problem.
Switched back to version 2.3 again, and it worked the first attempt just like the first time.
Just wanted to report this, in case anyone else was having an issue similar to mine.

gus33000, 2019-12-30:
Can you please clear the log file under \ProgramData\WPinternals\WPinternals.log, and re-attempt the unlock process using this build?

In case of failure I would need you to send the log file here for analysis from this version.
[WPinternals - 2.9.7303.14138.zip](https://github.com/ReneLergner/WPinternals/files/4009127/WPinternals.-.2.9.7303.14138.zip)

Thank you in advance.

gus33000, 2021-08-11:
Hi,

Are you able to try with the new version? We didn't hear back from you yet, we might have to close this issue as a result in a few days.

Thanks!

(New version: https://github.com/ReneLergner/WPinternals/releases/tag/2.9)

larsschellhas, 2022-09-26:
Hey @gus33000 , I'm actually facing the same issue at the moment.

Yesterday, I went through all unlock steps, however the phone was already unlocked. I didn't reflash it.

However, today, nothing seems to really work, so I flashed the original ffu, but no unlocking doesn't work.

I'll try again and give you the logs.

larsschellhas, 2022-09-26:
Here are the logs: 
[WPInternals.log](https://github.com/ReneLergner/WPinternals/files/9646708/WPInternals.log)

larsschellhas, 2022-09-26:
I tried to different PCs, two cables, and all USB ports I have :D

gus33000, 2024-10-12:
Please try using WPinternals 3.0 preview 3, if the issue persists, please file a new issue

closed 2024-10-12 by gus33000