ReneLergner/WPinternals #12: Failed to Write to Pipe

Empyreal96, 2020-05-07:
I'm trying to help someone unlock the bootloader of their 950, we tried version 2.8 and it was failing to find flash profile.
We used the FFU files downloaded with WDRT and WPInternals but still Failed to find flash profile as well as getting the error Failed to Write to Pipe..

We tried V2.4 as someone posted here saying that version worked for them, but still got Failed to write to pipe and Failed to find flash profile..

We can restore the device with WPInternals fine

Any suggestions? it's a 950 RM-1104

Thanks in advanced

michelinus, 2021-07-17:
For the people who have the same problem, a possible solution would be to remove the TF-Card, I also had the same problem and as soon as the TF-Card was removed the software was able to unlock the bootloader of my Lumia 950 RM-1104.

gus33000, 2021-08-11:
Hi, are you able to reproduce using the new version? https://github.com/ReneLergner/WPinternals/releases/tag/2.9

michelinus, 2021-08-12:
Hi, I had the lumia with only Windows 11 installed.

I tried to re-flash Windows 10 Mobile to re-unlock the bootloader but unfortunately the problem recurs even with WPInternals 2.9.2.

[image]

If needed I can provide the log file of the software to help debug the problem.

With the sd-card removed I can unlock the bootloader without problems.

XCraftTM, 2022-01-23:
Any Updates to this Problem? Becuase i have the same problem when i try to unlock the bootloader...

michelinus, 2022-01-23:
> Any Updates to this Problem? Becuase i have the same problem when i try to unlock the bootloader...

Have you tried removing the tf-card if you have one? Seems like the main issue

XCraftTM, 2022-01-23:
No i have no TF-Card in the Phone...

ShibeGG1, 2022-11-12:
What fixed it for me on my Lumia 930 is to take out the sim card tray (even if theres no sim in it)

ArcanjoMessiah, 2023-03-25:
Hi, i think i managed to solve the problem, first restore the bootloader and then with the phone in flash mode try unlocking the bootloader again. It worked on my Lumia 630

NoisyDream18, 2024-02-06:
hey none of this is working for me and i cant restore the bootloader because it goes tot he same error. Can anyone Identify the problem

closed 2024-10-12 by gus33000