ReneLergner/WPinternals #13: Lumia 535 - Mass Storage Mode new problem (v2.9 built today)

maniuraPL, 2020-07-12:
So, I've built the latest version of WPInternals as of today, it has different behaviour to release 2.8, in that version the bootloader of Lumia 535 could not be unlocked, after part 1 of unlocking WPInternals fails to find a flashing profile and the phone stays stuck in Flash mode, requiring a reflash to stock, as previously discovered and described. In this version, the bootloader gets unlocked as it should, goes into Mass Storage Mode for part 2, gets flashed, and gets back to Normal mode. After that, when switching into Mass Storage Mode, either manually or when performing any action requiring unlocked bootloader, the issue from 2.8 shows up, WPInternals fails to find a flashing profile, and the phone gets stuck in Flash mode, unable to switch to any other mode.

gus33000, 2021-08-11:
Hi, can you try the new version? https://github.com/ReneLergner/WPinternals/releases/tag/2.9

maniuraPL, 2021-08-11:
Tried it, results are exactly the same sadly, and may I add, even after a "successful" unlocking, all flags in thr Bootloader Security section of the Info tab are still red. In the Secure Boot technology section, UEFI and Effective Secure Boot status turn green.

gus33000, 2021-08-12:
> Tried it, results are exactly the same sadly, and may I add, even after a "successful" unlocking, all flags in the Bootloader Security section of the Info tab are still red. In the Secure Boot technology section, UEFI and Effective Secure Boot status turn green.

This is perfectly normal and expected for all Spec B Lumias, you won't ever get anything in the bootloader security section in green after the unlock, but it will still allow you to flash custom roms and boot unsigned code. (Refer to the getting started section of the tool to know more).

Daniel224455, 2023-03-13:
https://github.com/sonic011gamer/Lumia535Utils

Just use this guide, the WPinternals guide does NOT work.

closed 2024-10-12 by gus33000