ReneLergner/WPinternals #23: Flashing custom ffu hangs in "restoring bootloader"

mightycoco, 2021-06-10:
WPInternals 2.8

I downloaded 
Windows 10 Desktop (10.0.19043.962) (en-US) (Lumia 950 XL)
https://threedots.ovh/storage/windows/RM1085_1078.0053.1067.0000.2105.0022.19043.962_CLIENTPROFESSIONAL_a64fre_en-us_unsigned.ffu

Selected Flash / custom ffu

WPInternals is hanging in "Restoring bootloader..." - circle indicator is moving. The device has the red screen with the progress indicator at 0. I'm waiting since about 2 hours without anything changed.

mightycoco, 2021-06-10:
That's all in the WPInternals.log:

2021-06-10 08:28:42.853: Restoring bootloader...
2021-06-10 08:28:43.512: Rebooting phone
2021-06-10 08:28:43.545: Lumia disconnected
2021-06-10 08:28:49.687: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2021-06-10 08:28:49.687: Device path: \\?\USB#VID_045E&PID_0A02#6&359f9c5d&1&1#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2021-06-10 08:28:49.687: Connected device: Lumia
2021-06-10 08:28:49.688: Mode: Bootloader
2021-06-10 08:28:49.718: Bootmanager: 2.48
2021-06-10 08:28:49.719: Bootmanager protocol: 2.6
2021-06-10 08:28:49.719: Flash app: 2.75
2021-06-10 08:28:49.719: Flash protocol: 2.41
2021-06-10 08:28:49.719: SecureBoot: Disabled
2021-06-10 08:28:49.719: Flash app security: Disabled (FFU security: Disabled, RDC: Not found, Authenticated: False)
2021-06-10 08:28:49.719: JTAG: Enabled
2021-06-10 08:28:49.927: Flashing profile loaded

WPInternals.config:

<?xml version="1.0" encoding="utf-8"?>
<WPinternalsConfig>
  <FlashProfiles>
    <FlashProfile>
      <Type>RM-1085</Type>
      <PlatformID>Microsoft.MSM8994.P6211.2.1</PlatformID>
      <ProductCode>059X4V3</ProductCode>
      <PhoneFirmware>01078.00053.16236.35000</PhoneFirmware>
      <FfuFirmware>01078.00053.16236.35000</FfuFirmware>
      <Profile>AQAACAAAAAoAAw==</Profile>
    </FlashProfile>
  </FlashProfiles>
  <FFURepository>
    <FFUEntry>
      <PlatformID>Microsoft.MSM8994.P6211</PlatformID>
      <FirmwareVersion>01078.00053.16236.35000</FirmwareVersion>
      <OSVersion>10.0.10586.318</OSVersion>
      <Path>C:\ProgramData\WPinternals\Repository\RM-1085\RM1085_1078.0053.10586.13169.12547.035242_retail_prod_signed.ffu</Path>
    </FFUEntry>
    <FFUEntry>
      <PlatformID>Microsoft.MSM8994.P6211</PlatformID>
      <FirmwareVersion>01078.00053.16236.35000</FirmwareVersion>
      <OSVersion>10.0.10586.318</OSVersion>
      <Path>D:\etc\WPinternals 2.8\Repository\RM-1085\RM1085_1078.0053.10586.13169.12547.035242_retail_prod_signed.ffu</Path>
    </FFUEntry>
  </FFURepository>
  <EmergencyRepository>
    <EmergencyFileEntry>
      <Type>RM-1085</Type>
      <ProgrammerPath>C:\ProgramData\WPinternals\Repository\RM-1085\MPRG8994_fh.ede</ProgrammerPath>
      <PayloadPath>C:\ProgramData\WPinternals\Repository\RM-1085\RM1085_fh.edp</PayloadPath>
      <RKH>427D8FD5A7F227820D5B11BF8C6F7670C0A0622CC61BA95AAEE18F7517FC0B77</RKH>
    </EmergencyFileEntry>
    <EmergencyFileEntry>
      <Type>RM-1085</Type>
      <ProgrammerPath>D:\etc\WPinternals 2.8\Repository\RM-1085\MPRG8994_fh.ede</ProgrammerPath>
      <RKH>427D8FD5A7F227820D5B11BF8C6F7670C0A0622CC61BA95AAEE18F7517FC0B77</RKH>
    </EmergencyFileEntry>
  </EmergencyRepository>
</WPinternalsConfig>

gus33000, 2021-06-10:
Hi. WPinternals is not built to flash "custom"/unsigned ffu files. The place where you got the ffu should have instructed you to unlock the bootloader and flash using thor2 instead.

mediaexplorer74, 2023-07-15:
> https://threedots.ovh/storage/windows/RM1085_1078.0053.1067.0000.2105.0022.19043.962_CLIENTPROFESSIONAL_a64fre_en-us_unsigned.ffu

Gustave, https://threedots.ovh/storage/windows/RM1085_1078.0053.1067.0000.2105.0022.19043.962_CLIENTPROFESSIONAL_a64fre_en-us_unsigned.ffu not avalable any more. DId you have some big problems? I noticed that you heavy wiped all your own projects.... Too bad...  I cannot to probe 10X on my old sweet Lumia 950, heh. Ok. Have a nice time!
Best,
  m.e.

gus33000, 2023-07-15:
Hi,

What do you mean by heavy wipe? Nothing was removed at all on my side? The site you are begging about is not owned by me at all and never was...

Everything is still under the github.com/WOA-Project organisation and is not going away.

mediaexplorer74, 2023-07-15:
OK. I understand You (I see). I found that ffu... But.. 
I decide(d) to stay on W10M forever... because of possible MS's timebomb / drv problems.

joaozittoo, 2024-03-23:
**404 Not Found**

joaozittoo, 2024-03-23:
> > Hi. WPinternals is not built to flash "custom"/unsigned ffu files. The place where you got the ffu should have instructed you to unlock the bootloader and flash using thor2 instead.
> 
> Hello Gustave,
> 
> https://threedots.ovh/lumias/FFUs/RM1104_1078.0053.1067.0000.2102.210206-1800.2.20.20279.1002_modernpc_selfhost_a64fre_unsigned.ffu.7z not avalable any more. Did you have some big problems? I noticed that you heavy wiped all your own projects.... Too bad... I cannot probe to deploy your 10X to my old sweet Lumia 950, heh. Ok. Have a nice time!
> 
> Best, m.e.

404 Not Found

closed 2024-10-12 by gus33000