ReneLergner/WPinternals #25: Security checks seems to be in place after root

MarP13, 2021-08-13:
Hi guys, 

I went trough unlock and root of my Lumia 640 LTE on WinMo10 10.0.15254.547.

Now I am trying to benefit from that. But, when I try to write registry from UWP app, it fails on Unauthorized access. 
Alternatively, I tried to use PInvoke RegSetValueEx of forwarders\advapi32.dll. It return correct result code but change is not commited.

I tried other things too like using Process.Start to launch some processes (e.g. .NET core 3.1 self-hosted self-contained ARM console app) placed under Downloads folder. That failed on access denied as far as I rememeber. 

Is this normal? I am doing something wrong? 
Is there a way to verify if all system binaries are patched correctly?

Also, in the original youtube intro video posted by HeathCliff, he mentions that root enables you to run .NET console apps. Does anybody know how to prepare such a app?

Thanks a bunch for any information. I expect this is more of issue of my use&understanding of the root patch, but XDA-dev forum is dead so I am trying to find somebody knowledgeable here.

gus33000, 2021-08-14:
Hi, thanks for filing an issue on this repo!

Interop Tools seems to be able to access the registry without permission issues (and changes do stick), however this uses the RegistryRT library which makes use of Nt* calls for registry access. I recall older versions of WPinternals also had an SDK and a root access testing app, I'll dig this for you in a sec so you can tell me if both do seem to work.

gus33000, 2021-08-14:
Here's what I found:

[SDK.zip](https://github.com/ReneLergner/WPinternals/files/6986134/SDK.zip)
[Root Access Test App.zip](https://github.com/ReneLergner/WPinternals/files/6986135/Root.Access.Test.App.zip)

Let me know what this results in.

gus33000, 2021-08-14:
This is what the Test app uses by the way:

```
using System;
using System.Runtime.InteropServices;

namespace RootAccessTestUWP
{
	// Token: 0x02000004 RID: 4
	public static class Core
	{
		// Token: 0x0600000D RID: 13
		[DllImport("KERNELBASE.DLL", CharSet = 3, SetLastError = true)]
		private static extern int RegQueryValueEx(UIntPtr hKey, string lpValueName, int lpReserved, out Core.RegistryValueKind lpType, IntPtr lpData, ref int lpcbData);

		// Token: 0x0600000E RID: 14
		[DllImport("KERNELBASE.DLL", CharSet = 3)]
		private static extern int RegOpenKeyEx(UIntPtr hKey, string subKey, int ulOptions, int samDesired, out UIntPtr hkResult);

		// Token: 0x0600000F RID: 15
		[DllImport("KERNELBASE.DLL", SetLastError = true)]
		private static extern int RegCloseKey(UIntPtr hKey);

		// Token: 0x06000010 RID: 16 RVA: 0x000022E0 File Offset: 0x000004E0
		internal static string RegReadStringValue(UIntPtr rootKey, string keyPath, string valueName)
		{
			UIntPtr hKey;
			int num = Core.RegOpenKeyEx(rootKey, keyPath, 0, Core.KEY_READ, out hKey);
			if (num != 0)
			{
				throw new COMException("Failed to open registry key", num);
			}
			int num2 = 1024;
			IntPtr intPtr = Marshal.AllocHGlobal(num2);
			Core.RegistryValueKind registryValueKind;
			num = Core.RegQueryValueEx(hKey, valueName, 0, out registryValueKind, intPtr, ref num2);
			if (num != 0)
			{
				throw new COMException("Failed to read registry value", num);
			}
			if (registryValueKind != Core.RegistryValueKind.String)
			{
				throw new Exception("Wrong registry value type");
			}
			string result = Marshal.PtrToStringUni(intPtr);
			Core.RegCloseKey(hKey);
			return result;
		}

		// Token: 0x06000011 RID: 17 RVA: 0x00002358 File Offset: 0x00000558
		internal static uint RegReadDwordValue(UIntPtr rootKey, string keyPath, string valueName)
		{
			UIntPtr hKey;
			int num = Core.RegOpenKeyEx(rootKey, keyPath, 0, Core.KEY_READ, out hKey);
			if (num != 0)
			{
				throw new COMException("Failed to open registry key", num);
			}
			int num2 = 1024;
			IntPtr intPtr = Marshal.AllocHGlobal(num2);
			Core.RegistryValueKind registryValueKind;
			num = Core.RegQueryValueEx(hKey, valueName, 0, out registryValueKind, intPtr, ref num2);
			if (num != 0)
			{
				throw new COMException("Failed to read registry value", num);
			}
			if (registryValueKind != Core.RegistryValueKind.DWord)
			{
				throw new Exception("Wrong registry value type");
			}
			uint result = (uint)Marshal.ReadInt32(intPtr);
			Core.RegCloseKey(hKey);
			return result;
		}

		// Token: 0x04000005 RID: 5
		public static UIntPtr HKEY_CLASSES_ROOT = (UIntPtr)2147483648U;

		// Token: 0x04000006 RID: 6
		public static UIntPtr HKEY_CURRENT_USER = (UIntPtr)2147483649U;

		// Token: 0x04000007 RID: 7
		public static UIntPtr HKEY_LOCAL_MACHINE = (UIntPtr)2147483650U;

		// Token: 0x04000008 RID: 8
		public static UIntPtr HKEY_USERS = (UIntPtr)2147483651U;

		// Token: 0x04000009 RID: 9
		private static int KEY_READ = 131097;

		// Token: 0x0200000F RID: 15
		public enum RegistryValueKind
		{
			// Token: 0x0400002A RID: 42
			Unknown,
			// Token: 0x0400002B RID: 43
			String,
			// Token: 0x0400002C RID: 44
			ExpandString,
			// Token: 0x0400002D RID: 45
			Binary,
			// Token: 0x0400002E RID: 46
			DWord,
			// Token: 0x0400002F RID: 47
			MultiString = 7,
			// Token: 0x04000030 RID: 48
			QWord = 11,
			// Token: 0x04000031 RID: 49
			None = -1
		}
	}
}
```

gus33000, 2021-08-14:
I made available the source for both things on the repo just now as well if you want to take a look (here's the sdk: https://github.com/ReneLergner/WPinternals/tree/master/WPinternalsSDK)

MarP13, 2021-08-14:
Hi Gus, that was super helpful!
Seems that all the magic must be done via kernelbase.dll.  Registry editing works.
Environment.ShellExecute assumingly works as well, but I don't known how to redirect inputs yet.

What I noticed that cappability checks are most likely disable for UWPs too, because I was able to use this class:
https://docs.microsoft.com/en-us/uwp/api/windows.ui.input.preview.injection.inputinjector?view=winrt-20348

My goal is to autostart WIFI hotspot right after boot since I bought 640 (for $2) to serve as cheap LTE modem.
I was able to activate the hostpot via inputinject, now I need a way how to autostart the app.

gus33000, 2021-08-19:
Glad it worked out. i'm closing this issue but going to pin it to make it informative to other people. In the future it would be best however to have documentation on the repository

closed 2021-08-19 by gus33000