ReneLergner/WPinternals #27: Failed to write to pipe error when I want to unlock bootloader

ArixElo, 2021-08-19:
### Description

Hi, namely I want to unlock my Lumia 640, but when I try to do so, then it shows that error.

### Steps To Reproduce

I'm using RM-1085 donor FFU, and original FFU with the original Emergency Programmer.

### Expected behavior

[image]

### WPinternals Version

v2.8-v.2.9.2

### Windows Version

Windows 11 10.0.22200.132

### Phone model

Lumia 640 LTE  (RM-1072)

### Relevant Assets

2021-08-19 19:22:43.466: Windows Phone Internals version 2.9.7894.11970
2021-08-19 19:22:43.476: Copyright Heathcliff74 / wpinternals.net
2021-08-19 19:22:53.193: Flash App Type: FlashApp
2021-08-19 19:22:53.197: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2021-08-19 19:22:53.197: Device path: \\?\USB#VID_0421&PID_0714#5&21e739f&0&7#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2021-08-19 19:22:53.197: Connected device: Lumia
2021-08-19 19:22:53.197: Mode: Flash
2021-08-19 19:22:53.443: Phone type: RM-1072
2021-08-19 19:22:53.443: Product code: 059X1Q5
2021-08-19 19:22:53.444: Root key hash: 831BE6B18E7006372069545885A80F786C33D064904732D39EFBE0EC7019B5D4
2021-08-19 19:22:53.444: Firmware version: 02177.00000.15221.30000
2021-08-19 19:22:53.444: IMEI: 351858072581669
2021-08-19 19:22:53.444: Flash app: 2.60
2021-08-19 19:22:53.444: Flash protocol: 2.37
2021-08-19 19:22:53.444: SecureBoot: Enabled (Platform Secure Boot: Enabled, UEFI Secure Boot: Enabled)
2021-08-19 19:22:53.444: Flash app security: Enabled (FFU security: Enabled, RDC: Not found, Authenticated: False)
2021-08-19 19:22:53.444: JTAG: Disabled
2021-08-19 19:23:01.302: FFU not added, because it was already present in the repository.
2021-08-19 19:23:01.303: Emergency files not added, because they were already present in the repository.
2021-08-19 19:23:01.614: FFU not added, because it was already present in the repository.
2021-08-19 19:23:01.614: Unlock Bootloader
2021-08-19 19:23:01.614: Processing resources:
2021-08-19 19:23:01.614: Profile FFU: D:\ścierwo\Lumia 640 stuff\WPinternals 2.8\RM1072_02177.00000.15221.30005_RETAIL_prod_signed_1022_026134_ORG-PL(1).ffu
2021-08-19 19:23:01.615: EDE file: C:\Program Files (x86)\Microsoft Care Suite\Windows Device Recovery Tool\MPRG8x26_fh.ede
2021-08-19 19:23:01.615: Donor-FFU with supported OS version: D:\ścierwo\Lumia 640 stuff\WPinternals 2.8\RM1085_1078.0053.10586.13169.12584.034D55_retail_prod_signed.ffu
2021-08-19 19:23:03.368: Assembling data for unlock
2021-08-19 19:23:03.369: Assembling data for unlock
2021-08-19 19:23:03.440: Attempt patch: SecureBootHack-V2-EFIESP
2021-08-19 19:23:03.446: Pattern: 0, 0
2021-08-19 19:23:03.446: Pattern: 1, 0
2021-08-19 19:23:03.446: Pattern: 2, 0
2021-08-19 19:23:03.446: Pattern: 3, 0
2021-08-19 19:23:03.446: Pattern: 4, 0
2021-08-19 19:23:03.446: Pattern: 5, 0
2021-08-19 19:23:03.446: Pattern: 6, 0
2021-08-19 19:23:03.446: Pattern: 7, 0
2021-08-19 19:23:03.446: Pattern: 8, 0
2021-08-19 19:23:03.446: Pattern: 9, 0
2021-08-19 19:23:03.446: Pattern: 10, 0
2021-08-19 19:23:03.447: Pattern: 11, 0
2021-08-19 19:23:03.447: Pattern: 12, 0
2021-08-19 19:23:03.447: Pattern: 13, 0
2021-08-19 19:23:03.447: Pattern: 14, 0
2021-08-19 19:23:03.447: Pattern: 15, 0
[... 519 lines cut ...]
2021-08-19 19:25:06.757: Error: Failed to write to pipe: 4E4F4B58464645
2021-08-19 19:25:06.757:     Error: Failed to write pipe on WinUSB device.
2021-08-19 19:25:06.757:        at MadWizard.WinUSBNet.API.WinUSBDevice.WritePipe(Int32 ifaceIndex, Byte pipeID, Byte[] buffer, Int32 offset, Int32 length)
   at MadWizard.WinUSBNet.USBPipe.Write(Byte[] buffer, Int32 offset, Int32 length)
2021-08-19 19:25:06.757:         Error: Urzdzenie doczone do komputera nie dziaa.
2021-08-19 19:25:06.759: Error: Failed to write to pipe: 4E4F4B58464645
2021-08-19 19:25:06.762:    at MadWizard.WinUSBNet.USBPipe.LogAndThrowException(Exception Ex)
   at MadWizard.WinUSBNet.USBPipe.Write(Byte[] buffer, Int32 offset, Int32 length)
   at WPinternals.NokiaPhoneModel.ExecuteRawMethod(Byte[] RawMethod, Int32 Length)
   at WPinternals.NokiaPhoneModel.ExecuteRawMethod(Byte[] RawMethod)
   at WPinternals.NokiaFlashModel.EndAsyncFlash()
   at WPinternals.LumiaV2UnlockBootViewModel.LumiaV2CustomFlashInternal(PhoneNotifierViewModel Notifier, String FFUPath, Boolean PerformFullFlashFirst, Boolean SkipWrite, List`1 FlashParts, Boolean DoResetFirst, Boolean ClearFlashingStatusAtEnd, Boolean CheckSectorAlignment, Boolean ShowProgress, Boolean Experimental, SetWorkingStatus SetWorkingStatus, UpdateWorkingStatus UpdateWorkingStatus, ExitSuccess ExitSuccess, ExitFailure ExitFailure, String ProgrammerPath)
   at WPinternals.LumiaV2UnlockBootViewModel.LumiaV2CustomFlash(PhoneNotifierViewModel Notifier, String FFUPath, Boolean PerformFullFlashFirst, Boolean SkipWrite, List`1 FlashParts, Boolean DoResetFirst, Boolean ClearFlashingStatusAtEnd, Boolean CheckSectorAlignment, Boolean ShowProgress, Boolean Experimental, SetWorkingStatus SetWorkingStatus, UpdateWorkingStatus UpdateWorkingStatus, ExitSuccess ExitSuccess, ExitFailure ExitFailure, String ProgrammerPath)
   at WPinternals.LumiaUnlockBootloaderViewModel.LumiaFlashParts(PhoneNotifierViewModel Notifier, String FFUPath, Boolean PerformFullFlashFirst, Boolean SkipWrite, List`1 Parts, Boolean DoResetFirst, Boolean ClearFlashingStatusAtEnd, Boolean CheckSectorAlignment, Boolean ShowProgress, Boolean Experimental, SetWorkingStatus SetWorkingStatus, UpdateWorkingStatus UpdateWorkingStatus, ExitSuccess ExitSuccess, ExitFailure ExitFailure, String EDEPath)
   at WPinternals.LumiaUnlockBootloaderViewModel.LumiaUnlockUEFI(PhoneNotifierViewModel Notifier, String ProfileFFUPath, String EDEPath, String SupportedFFUPath, SetWorkingStatus SetWorkingStatus, UpdateWorkingStatus UpdateWorkingStatus, ExitSuccess ExitSuccess, ExitFailure ExitFailure, Boolean ExperimentalSpecBEFIESPUnlock, Boolean ExperimentalSpecAEFIESPUnlock, Boolean ReUnlockDevice)
2021-08-19 19:25:06.762:     Error: Failed to write pipe on WinUSB device.
2021-08-19 19:25:06.762:        at MadWizard.WinUSBNet.API.WinUSBDevice.WritePipe(Int32 ifaceIndex, Byte pipeID, Byte[] buffer, Int32 offset, Int32 length)
   at MadWizard.WinUSBNet.USBPipe.Write(Byte[] buffer, Int32 offset, Int32 length)
2021-08-19 19:25:06.762:         Error: Urzdzenie doczone do komputera nie dziaa.
2021-08-19 19:25:06.763: Failed to write to pipe: 4E4F4B58464645
2021-08-19 19:25:07.145: Lumia disconnected
2021-08-19 19:25:07.151: Found device on interface: 71de994d-8b7c-43db-a27e-2ae7cd579a0c
2021-08-19 19:25:07.151: Device path: \\?\USB#VID_05C6&PID_9008#5&21e739f&0&7#{71de994d-8b7c-43db-a27e-2ae7cd579a0c}
2021-08-19 19:25:07.151: Connected device: Lumia
2021-08-19 19:25:07.151: Mode: Qualcomm Emergency Download 9008

gus33000, 2021-08-19:
Just as a note, you wouldn't be running this under a virtual machine? WPinternals does not work well under VMs...

ArixElo, 2021-08-19:
> Just as a note, you wouldn't be running this under a virtual machine? WPinternals does not work well under VMs...

Nope, I'm running on real hardware.

xordos, 2021-08-22:
I have similar error

Lumia 640 US ATT SIM locked, I have tried numerous times, so it past the attempt profile phase, and when try to unlock, the latest log is following:

2021-08-21 19:44:45.699: Lumia disconnected
2021-08-21 19:45:01.279: Found device on interface: 08324f9c-b621-435c-859b-ae4652481b7c
2021-08-21 19:45:01.279: Device path: \\?\USB#VID_0421&PID_06FC&MI_03#7&15472169&0&0003#{08324f9c-b621-435c-859b-ae4652481b7c}
2021-08-21 19:45:01.279: Connected device: Lumia
2021-08-21 19:45:01.279: Mode: Normal
2021-08-21 19:45:01.283: Failed to switch to Mass Storage Mode
2021-08-21 19:45:01.316: Operator: ATT-US
2021-08-21 19:45:01.318: Manufacturer Model Name: RM-1073_1004
2021-08-21 19:45:01.332: Product Code: 059X0B8
2021-08-21 19:45:01.334: Firmware: 02177.00000.15184.36016
2021-08-21 19:45:01.337: HWID: 2140
2021-08-21 19:45:01.347: IMEI: 3xxx_masked
2021-08-21 19:45:01.350: Public ID: F6 26 54 AC 57 18 AF D4 68 B0 8B A1 57 89 09 4B 70 51 36 89
2021-08-21 19:45:01.353: Bluetooth MAC: 30 0D 43 48 05 86
2021-08-21 19:45:01.359: WLAN MAC 1: 30 0D 43 57 D4 64
2021-08-21 19:45:01.369: WLAN MAC 2: 30 0D 43 57 D4 65
2021-08-21 19:45:01.376: WLAN MAC 3: 30 0D 43 57 D4 66
2021-08-21 19:45:01.383: WLAN MAC 4: 30 0D 43 57 D4 67
2021-08-21 19:45:01.398: Bootloader Security: Enabled
2021-08-21 19:45:01.411: Simlock: Active
2021-08-21 19:45:01.486: Boot policy: Unsecure
2021-08-21 19:45:01.486: DB: SecureProduction
2021-08-21 19:45:01.486: DBX: Unknown
2021-08-21 19:45:01.486: KEK: SecureProduction
2021-08-21 19:45:01.486: PK: SecureProduction
2021-08-21 19:46:35.833: Rebooting phone to Flash mode
2021-08-21 19:46:35.833: Rebooting phone to Flash mode...
2021-08-21 19:46:50.643: Lumia disconnected
2021-08-21 19:46:56.645: Flash App Type: FlashApp
2021-08-21 19:46:56.648: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2021-08-21 19:46:56.649: Device path: \\?\USB#VID_0421&PID_0714#6&3af0f9ce&0&14#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2021-08-21 19:46:56.649: Connected device: Lumia
2021-08-21 19:46:56.649: Mode: Flash
2021-08-21 19:46:56.725: Security flags: 0x00007BFF
2021-08-21 19:46:57.026: Platform Name: Nokia.MSM8926.P6204_ATT.1.1
2021-08-21 19:46:57.043: Public ID: F6 26 54 AC 57 18 AF D4 68 B0 8B A1 57 89 09 4B 70 51 36 89
2021-08-21 19:46:57.063: Root Key Hash: 83 1B E6 B1 8E 70 06 37 20 69 54 58 85 A8 0F 78 6C 33 D0 64 90 47 32 D3 9E FB E0 EC 70 19 B5 D4
2021-08-21 19:46:57.063: Platform Secure Boot Status: True
2021-08-21 19:46:57.063: Uefi Secure Boot Status: False
2021-08-21 19:46:57.063: Effective Secure Boot Status: False
2021-08-21 19:46:57.063: Bootloader Security Qfuse Status: True
2021-08-21 19:46:57.063: Bootloader Security Authentication Status: False
2021-08-21 19:46:57.063: Bootloader Security Rdc Status: False
2021-08-21 19:46:57.063: Effective Bootloader Security Status: True
2021-08-21 19:46:57.063: Native Debug Status: False
2021-08-21 19:46:58.231: Charging status: 92% - 210 mA (charging)
2021-08-21 19:46:58.415: Phone type: RM-1073
2021-08-21 19:46:58.415: Product code: 059X0B8
2021-08-21 19:46:58.415: Root key hash: 831BE6B18E7006372069545885A80F786C33D064904732D39EFBE0EC7019B5D4
2021-08-21 19:46:58.415: Firmware version: 02177.00000.15184.36000
2021-08-21 19:46:58.415: IMEI: 3xxx_masked
2021-08-21 19:46:58.415: Flash app: 2.60
2021-08-21 19:46:58.415: Flash protocol: 2.37
2021-08-21 19:46:58.415: SecureBoot: Disabled (Platform Secure Boot: Enabled, UEFI Secure Boot: Enabled)
2021-08-21 19:46:58.415: Flash app security: Enabled (FFU security: Enabled, RDC: Not found, Authenticated: False)
2021-08-21 19:46:58.415: JTAG: Disabled
2021-08-21 19:46:58.415: Bootloader: Lumia Bootloader Spec B
2021-08-21 19:46:58.415: ProductCode: 059X0B8
2021-08-21 19:46:58.415: ProductType: RM-1073
2021-08-21 19:54:33.673: FFU not added, because it was already present in the repository.
2021-08-21 19:54:33.673: Emergency files not added, because they were already present in the repository.
2021-08-21 19:54:33.730: FFU not added, because it was already present in the repository.
2021-08-21 19:54:33.730: Unlock Bootloader
2021-08-21 19:54:33.730: Processing resources:
2021-08-21 19:54:33.730: Profile FFU: C:\ProgramData\WPInternals\R3\RM-1073\RM1073_02177.00000.15184.36016_RETAIL_prod_signed_1004_022B79_ATT-US.ffu
2021-08-21 19:54:33.730: EDE file: C:\ProgramData\WPInternals\R3\RM-1073\MPRG8x26_fh.ede
[... 67 lines cut ...]
2021-08-21 19:57:42.787: Failed to write to pipe: 4E4F4B58434242
2021-08-21 19:57:42.793: Lumia disconnected
2021-08-21 19:57:58.194: Found device on interface: 08324f9c-b621-435c-859b-ae4652481b7c
2021-08-21 19:57:58.194: Device path: \\?\USB#VID_0421&PID_06FC&MI_03#7&15472169&0&0003#{08324f9c-b621-435c-859b-ae4652481b7c}
2021-08-21 19:57:58.194: Connected device: Lumia
2021-08-21 19:57:58.194: Mode: Normal
2021-08-21 19:57:58.218: Operator: ATT-US
2021-08-21 19:57:58.219: Manufacturer Model Name: RM-1073_1004
2021-08-21 19:57:58.233: Product Code: 059X0B8
2021-08-21 19:57:58.235: Firmware: 02177.00000.15184.36016
2021-08-21 19:57:58.238: HWID: 2140
2021-08-21 19:57:58.246: IMEI: 3xxx_masked
2021-08-21 19:57:58.249: Public ID: F6 26 54 AC 57 18 AF D4 68 B0 8B A1 57 89 09 4B 70 51 36 89
2021-08-21 19:57:58.252: Bluetooth MAC: 30 0D 43 48 05 86
2021-08-21 19:57:58.258: WLAN MAC 1: 30 0D 43 57 D4 64
2021-08-21 19:57:58.264: WLAN MAC 2: 30 0D 43 57 D4 65
2021-08-21 19:57:58.270: WLAN MAC 3: 30 0D 43 57 D4 66
2021-08-21 19:57:58.276: WLAN MAC 4: 30 0D 43 57 D4 67
2021-08-21 19:57:58.284: Bootloader Security: Enabled
2021-08-21 19:57:58.296: Simlock: Active
2021-08-21 19:57:58.356: Boot policy: Unsecure
2021-08-21 19:57:58.356: DB: SecureProduction
2021-08-21 19:57:58.356: DBX: Unknown
2021-08-21 19:57:58.356: KEK: SecureProduction
2021-08-21 19:57:58.356: PK: SecureProduction

xordos, 2021-08-22:
Another scenarios (I just tried again after above failure), go to flash mode, make sure 3 files selected, and press unlock again, this time, it says something flashed successfully, then display "signing" for about 3 min, then display error with different value "Failed to write to pipe: 4E4F4B56"

2021-08-21 19:57:58.356: KEK: SecureProduction
2021-08-21 19:57:58.356: PK: SecureProduction
2021-08-21 20:12:26.423: Start unlock. Phone needs to switch to Flash-mode
2021-08-21 20:12:26.423: In order to start unlocking the bootloader, the phone needs to be switched to Flash-mode.
2021-08-21 20:12:27.703: Switching to Flash-mode
2021-08-21 20:12:27.703: Rebooting phone to Flash mode
2021-08-21 20:12:27.703: Rebooting phone to Flash mode...
2021-08-21 20:12:43.200: Lumia disconnected
2021-08-21 20:12:48.821: Flash App Type: FlashApp
2021-08-21 20:12:48.825: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2021-08-21 20:12:48.825: Device path: \\?\USB#VID_0421&PID_0714#6&3af0f9ce&0&14#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2021-08-21 20:12:48.825: Connected device: Lumia
2021-08-21 20:12:48.825: Mode: Flash
2021-08-21 20:12:49.049: Phone type: RM-1073
2021-08-21 20:12:49.049: Product code: 059X0B8
2021-08-21 20:12:49.049: Root key hash: 831BE6B18E7006372069545885A80F786C33D064904732D39EFBE0EC7019B5D4
2021-08-21 20:12:49.049: Firmware version: 02177.00000.15184.36000
2021-08-21 20:12:49.049: IMEI: masked
2021-08-21 20:12:49.049: Flash app: 2.60
2021-08-21 20:12:49.049: Flash protocol: 2.37
2021-08-21 20:12:49.049: SecureBoot: Disabled (Platform Secure Boot: Enabled, UEFI Secure Boot: Enabled)
2021-08-21 20:12:49.049: Flash app security: Enabled (FFU security: Enabled, RDC: Not found, Authenticated: False)
2021-08-21 20:12:49.049: JTAG: Disabled
2021-08-21 20:12:59.308: FFU not added, because it was already present in the repository.
2021-08-21 20:12:59.308: Emergency files not added, because they were already present in the repository.
2021-08-21 20:12:59.363: FFU not added, because it was already present in the repository.
2021-08-21 20:12:59.363: Unlock Bootloader
2021-08-21 20:12:59.363: Processing resources:
2021-08-21 20:12:59.363: Profile FFU: C:\ProgramData\WPInternals\R3\RM-1073\RM1073_02177.00000.15184.36016_RETAIL_prod_signed_1004_022B79_ATT-US.ffu
2021-08-21 20:12:59.363: EDE file: C:\ProgramData\WPInternals\R3\RM-1073\MPRG8x26_fh.ede
2021-08-21 20:12:59.363: Donor-FFU with supported OS version: C:\ProgramData\WPInternals\R3\RM-1085\RM1085_1078.0053.10586.13169.12547.035242_retail_prod_signed.ffu
2021-08-21 20:12:59.620: Assembling data for unlock
2021-08-21 20:12:59.620: Assembling data for unlock
2021-08-21 20:12:59.631: Attempt patch: SecureBootHack-V2-EFIESP
2021-08-21 20:12:59.632: Pattern: 0, 0
2021-08-21 20:12:59.632: Pattern: 1, 0
2021-08-21 20:12:59.632: Pattern: 2, 0
2021-08-21 20:12:59.632: Pattern: 3, 0
2021-08-21 20:12:59.632: Pattern: 4, 0
2021-08-21 20:12:59.632: Pattern: 5, 0
2021-08-21 20:12:59.633: Pattern: 6, 0
2021-08-21 20:12:59.633: Pattern: 7, 0
2021-08-21 20:12:59.633: Pattern: 8, 0
2021-08-21 20:12:59.633: Pattern: 9, 0
2021-08-21 20:12:59.633: Pattern: 10, 0
2021-08-21 20:12:59.633: Pattern: 11, 0
2021-08-21 20:12:59.633: Pattern: 12, 0
2021-08-21 20:12:59.633: Pattern: 13, 0
2021-08-21 20:12:59.633: Pattern: 14, 0
2021-08-21 20:12:59.633: Pattern: 15, 0
2021-08-21 20:12:59.633: Pattern: 16, 0
2021-08-21 20:12:59.633: Pattern: 17, 0
2021-08-21 20:12:59.633: Pattern: 18, 0
2021-08-21 20:12:59.633: Pattern: 19, 0
2021-08-21 20:12:59.633: Pattern: 20, 0
2021-08-21 20:12:59.633: Donor-FFU: C:\ProgramData\WPInternals\R3\RM-1085\RM1085_1078.0053.10586.13169.12547.035242_retail_prod_signed.ffu
2021-08-21 20:12:59.645: Taking mobilestartup.efi from donor-FFU
2021-08-21 20:12:59.645: Attempt patch: SecureBootHack-V2-EFIESP
2021-08-21 20:12:59.648: Pattern: 0, 0
2021-08-21 20:12:59.648: Pattern: 1, 0
2021-08-21 20:12:59.648: Pattern: 2, 0
2021-08-21 20:12:59.648: Pattern: 3, 0
2021-08-21 20:12:59.648: Pattern: 4, 0
2021-08-21 20:12:59.648: Apply: 10.0.10586.318
2021-08-21 20:12:59.648: Edit BCD
2021-08-21 20:12:59.681: Enabling Test Signing
2021-08-21 20:12:59.681: Enabling Test Signing
2021-08-21 20:12:59.935: Initializing flash...
[... 102 lines cut ...]
2021-08-21 20:16:16.050: Mode: Bootloader
2021-08-21 20:16:26.133: Lumia disconnected
2021-08-21 20:16:41.406: Found device on interface: 08324f9c-b621-435c-859b-ae4652481b7c
2021-08-21 20:16:41.406: Device path: \\?\USB#VID_0421&PID_06FC&MI_03#7&15472169&0&0003#{08324f9c-b621-435c-859b-ae4652481b7c}
2021-08-21 20:16:41.406: Connected device: Lumia
2021-08-21 20:16:41.406: Mode: Normal
2021-08-21 20:16:41.427: Operator: ATT-US
2021-08-21 20:16:41.429: Manufacturer Model Name: RM-1073_1004
2021-08-21 20:16:41.446: Product Code: 059X0B8
2021-08-21 20:16:41.447: Firmware: 02177.00000.15184.36016
2021-08-21 20:16:41.451: HWID: 2140
2021-08-21 20:16:41.470: IMEI: masked
2021-08-21 20:16:41.474: Public ID: F6 26 54 AC 57 18 AF D4 68 B0 8B A1 57 89 09 4B 70 51 36 89
2021-08-21 20:16:41.477: Bluetooth MAC: 30 0D 43 48 05 86
2021-08-21 20:16:41.485: WLAN MAC 1: 30 0D 43 57 D4 64
2021-08-21 20:16:41.493: WLAN MAC 2: 30 0D 43 57 D4 65
2021-08-21 20:16:41.499: WLAN MAC 3: 30 0D 43 57 D4 66
2021-08-21 20:16:41.506: WLAN MAC 4: 30 0D 43 57 D4 67
2021-08-21 20:16:41.514: Bootloader Security: Enabled
2021-08-21 20:16:41.522: Simlock: Active
2021-08-21 20:16:41.590: Boot policy: Unsecure
2021-08-21 20:16:41.590: DB: SecureProduction
2021-08-21 20:16:41.590: DBX: Unknown
2021-08-21 20:16:41.590: KEK: SecureProduction
2021-08-21 20:16:41.590: PK: SecureProduction

xordos, 2021-08-22:
Want report back my progress.  I found another issue mentioned 2.3 works better. so I tried that, and it indeed worked.

I have include the 2.3 log including the download URL in the log.  Hopefully this can help gus and others.  And thanks for creating and continue maintaining this tool!

[WPI_v2.3_log.txt](https://github.com/ReneLergner/WPinternals/files/7027882/WPI_v2.3_log.txt)

ArixElo, 2021-08-22:
> Want report back my progress. I found another issue mentioned ver 2.3 works better. so I tried that, and it indeed worked.
> 
> I have include the 2.3 log including the download URL in the log. Hopefully this can help gus and others. And thanks for creating and continue maintaining this tool!
> 
> [WPI_v2.3_log.txt](https://github.com/ReneLergner/WPinternals/files/7027882/WPI_v2.3_log.txt)

Hi, I just used the same version as you, and it works! Thank you.

hooddy, 2021-12-17:
I have the same with my L830. Strange that its 1st time ive faced with it. Previously its unlocked succefuly. I dunno the reasons.

ArixElo, 2021-12-17:
> I have the same with my L830. Strange that its 1st time ive faced with it. Previously its unlocked succefuly. I dunno the reasons.

Hi, I don't know why it didn't worked at 2.8, as I said before: I tried with 2.3 and it worked straight away, so I'm closing this issue.

closed 2021-12-17 by ArixElo

hooddy, 2021-12-17:
Unfortunately WP2.3 to me works with the same error.

raymondchalme, 2023-12-05:
I'm running into a very similar issue here but it appears that v2.3 is no longer available?