ReneLergner/WPinternals #30: Unlocking Bootloader is stuck on Error 0x1106: Security header validation failed

t-mxcom, 2021-09-24:
### Description

I followed this guide: https://woa-project.github.io/LumiaWOA/guides/unlock/
The last step before "We're done" resulted in the error **Error 0x1106: Security header validation failed** and since then I cannot do anything with the phone.
[image]

Trying to boot the old OS results in a Bitlocker being displayed which wants me to provide an USB stick with  recovery files.

Also the Windows Device Recovery Tool is not able to do anything.
After it detects the phone, it states that there is no space on the disk (sorry for the screenshot being in German).
[image]

Find the `WPInternals.log` attached.

### Steps To Reproduce

1. See link in description. I can't add more details.

### Expected behavior

At the moment I would be happy being able to flash the original FFU back to the device.

THANK YOU VERY MUCH FOR YOUR SUPPORT!

### WPinternals Version

2.9.7894.11970

### Windows Version

Edition	Windows 10 Pro Version	21H1 Installiert am	‎26.‎06.‎2020 Betriebssystembuild	19043.1237 Leistung	Windows Feature Experience Pack 120.2212.3530.0

### Phone model

Lumia 950 XL (RM-1085)

### Relevant Assets

[WPInternals.log](https://github.com/ReneLergner/WPinternals/files/7226949/WPInternals.log)

gus33000, 2021-11-12:
Hi sorry for the delay, are you still having issues with the device and do you still have it?

maliksantika, 2021-11-13:
I also face the same case.
Can you teach a tutorial?

gus33000, 2021-11-13:
Put the phone in flash mode
Close WPinternals
Position yourself in a folder with the thor2.exe tool (it's included with WDRT)

Run this command:
```thor2 -mode rnd -boot_edmode```

Assuming you used WPinternals download function, you should have these paths matching except for model, if not, you'll have to download manually from lumiafirmware:

```thor2 -mode emergency -hexfile  C:\ProgramData\WPInternals\Repository\RM-1104\MPRG8992_fh.ede -edfile C:\ProgramData\WPInternals\Repository\RM-1104\RM1104_fh.edp```

Replace RM1104 and RM-1104 and 8992 with what matches for your phone, you can get both files here: https://www.lumiafirmware.com/ (just input your product code and download them in the emergency section.

Then you should be able to flash ffus

To recap:

1. Click "interrupt the boot sequence" in wpinternals to go to flash app
2. reboot the device and keep it connected
3. CLOSE wpinternals once flash app appears (arrows at the top on the phone screen)
4. open command prompt
5. cd c:\Program Files (x86)\Microsoft Care Suite\Windows Device Recovery Tool
6. thor2 -mode rnd -boot_edmode
7. thor2 -mode emergency -hexfile  C:\ProgramData\WPInternals\Repository\RM-1085\MPRG8994_fh.ede -edfile C:\ProgramData\WPInternals\Repository\RM-1085\RM1085_fh.edp
8. close command prompt
9. Use WDRT

maliksantika, 2021-11-13:
hallo sirrrrr

You are amazingaaaaaaa

salute

Thank you very muchkk

Salute

Headquarter  ARCAMANIK  ESTATE
Jl. Pacuan Kuda No 04 Sukamiskin Arcamanik Bandung 40293
Jawa Barat - Indonesia

☎ +62 22-63739188
☎ +62 11-200 4846
☎ +62 878-1009-5630

maliksantika, 2021-11-13:
thank youu sirr

t-mxcom, 2022-02-13:
Hi,

i just tried the tutorial you wrote on the 13th of November and everything looked fine until step 7.

Step 6 ended displaying the messages:
```
Device rebooted to ED mode successfully.
Exited with success
```
(see [thor2 -mode rnd -boot_edmode.log](https://github.com/ReneLergner/WPinternals/files/8056530/thor2.-mode.rnd.-boot_edmode.log))

So I ran step 7 which unfortunately resulted in displaying:
```
Clearing the backup GPT...SKIPPED!
Unable to parse FFU file. File open failed
programming operation failed!
UEFI FLASH END
Operation took about 15.00 seconds.

FFU_PARSING_ERROR

THOR2 1.8.2.18 exited with error code 2228224 (0x220000)
```
(see [thor2 -mode emergency.log](https://github.com/ReneLergner/WPinternals/files/8056529/thor2.-mode.emergency.log))

After that, the phone displayed the arrows on red background but I couldn't get any thor2-command to connect anymore.
Everytime it tried to connect, the following lines showed on the screen:
```
Detecting UEFI responder
Send HELLO
Send HELLO
Send HELLO
Send HELLO
Send HELLO
Lumia UEFI Application did not respond to version info query
Device is not in Lumia UEFI mode
```

The connection list showed:
```
WinUSB in use.
Connection list START
0.1D00:0010:0002        {9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}   Flash mode connected
Connection list END

Exited with success
```

As nothing worked, I decided to remove the battery and put it back in.
Now - even worse - it seems that the phone is completely dead.
I cannot get it to power on, connect or do anything.
Seems like it was a quite stupid decision! :(

Do you have any ideas that could bring my phone back to life?

THANK YOU VERY MUCH in advance!

gus33000, 2022-03-15:
@t-mxcom Hi sorry for the delay in replying (I'm not for some reason getting notifications on this repo, but I don't exactly own it so that might be why)

This error was expected, you just had to close and open WDRT. If the phone doesn't turn on now it means the battery is empty, you have to charge it externally

TheDomcio, 2022-03-16:
> Put the phone in flash mode Close WPinternals Position yourself in a folder with the thor2.exe tool (it's included with WDRT)
>
> Run this command: `thor2 -mode rnd -boot_edmode`
>
> Assuming you used [...]

Thank you very much for this solution friend. I've tried many methods for many hours (fortunately I didn't repeat the story with my Lumia 1020 64GB Edition, where I must have searched for about two months for a method to solve the problem and finally the problem turned out to be the official firmware, which for some reason didn't want to download the latest update from the server, so I uploaded a twin rom from another country), but this soluition turned out to be the only one. I've bricked my Lumia 950XL (RM-1116 - 059X523 and RKH: 427D8FD5A7F227820D5B11BF8C6F7670C0A0622CC61BA95AAEE18F7517FC0B77 - I'm providing this data for search engines to index, as if someone was ever looking for a solution) when accidentally disconnecting the charging cable while unlocking the bootlader.

MorphedBox, 2022-07-19:
> Put the phone in flash mode Close WPinternals Position yourself in a folder with the thor2.exe tool (it's included with WDRT)
>
> Run this command: `thor2 -mode rnd -boot_edmode`
>
> Assuming you used [...]

YOU DID IT!! You are the reason I finally unbricked my Lumia 950 XL after all these years!

I am beyond thankful!! <3 Hope you have the greatest day ever!

(P.S. I created a github just to thank you :)

infevlol, 2022-07-21:
THANK YOU!!! You helped me unbrick my Lumia 635 after all these months!!!
Hope you have the greatest life anybody on this planet has ever had!!! <3333

dgsolve, 2022-09-18:
Dear Sir Gus330000!!
The immortal words of Luis Armstrong come to mind following your kind tutorial.
What a wonderful world indeed 😁❤️❤️❤️❤️!!!
Thank you.
from a now un-looped 950XL

HOldenburg, 2022-10-11:
You bricked the 950xl? did you try the OEM reloader? I recovered mine - but I don't know what your symptoms are - The Windows Device Recovery Tool saved mine and I was able to unlock the bootloader with the other utility.

> Thank you very much for this solution friend. I've tried many methods for many hours (fortunately I didn't repeat the story with my Lumia 1020 64GB Edition, where I must have searched for about two [...]

Best Regards;

Herb
***@***.***

t-mxcom, 2022-12-04:
Hello,

I needed some time to charge the battery which i finally did by connecting wires of a USB cable directly to the connectors of the battery. After charging for about 1-2 hours it had enough energy to power up the phone to the red screen with white arrows.

Running in that state, the Windows Device Recovery Tool was able to install the software!

THANK YOU VERY VERY MUCH FOR YOUR HELP!

Greetings from Austria,
Max

wandelreyof, 2023-10-24:
I deleted the partitions on my Lumia 950xl, does anyone know, any tutorial to recover the cell phone, as it doesn't even turn on anymore

Dinico414, 2024-01-03:
@gus33000 you´re the best!!! but is there also a way to do this for a lumia which does only black screen no vibration (and no battery is not empty xD)

closed 2024-10-12 by gus33000