ReneLergner/WPinternals #34: Interest in WP7.x devices.

thenxguy, 2021-11-01:
Hey everyone!
With the WP7 marketplace being down for years, these phones like the Lumia 610 are only calling and texting devices. I'd be willing to get deep into "hacking" the 610 and maybe others if I could get them. @gus33000 How would I start with this? Is there a reason why you guys didn't support those devices?
Best regards

gus33000, 2021-11-01:
WPinternals exploits are built around a component only present on Lumia Spec A or higher devices called FlashApp. The tool is so built around this that "adding support for WP7 devices" is not only not possible with any of the existing code, it's also very out of scope for this project. Not even the binary patches will be compatible here, we're talking UEFI + NT versus Custom CE loader + WinCE

gus33000, 2021-11-01:
This would warrant an entirely new project, but that's up to you, we don't have any knowledge for WP7 and don't have the time to handle this. There were solutions available for "rooting WP7" in the past that are software and firmware based, look around XDA for those. But that's the end of my knowledge here.

closed 2021-11-01 by gus33000

thenxguy, 2021-11-01:
Thanks for the explanation. Dang it, I'll have to figure it out myself. But that is interesting. I've been mislead to think that CE was dead after Mobile 6. Have a great day.