ReneLergner/WPinternals #35: Application freezes while controlling device

lgh-127001, 2022-01-11:
Hello,
I'm using WPInternals 2.9.2 downloaded from the releases page. I have a Nokia Lumia 1320 (Windows 8.1) which I would like to root for deploying applications written by me. I connect the phone via USB, start the application, the "info" panel displays the Lumia information, detects that the phone is running in normal mode, but product code, etc. fields are empty. (The phone is turned on and reacts normally for everything, so I can unlock the screen, etc and both the application and Windows detects it - the application shows normal boot mode, the Windows shows SD card content). But when I click on "flash mode" the application window becomes unresponsive: doesn't react to anything, I can't switch between modes of the application (for example can't switch to "Getting started" window), the app doesn't react to anything. In the log I see only a single message related to this apart from the normal version and such informations:
2022-01-11 08:45:40.733: Rebooting phone to Flash mode
After that, nothing. I waited 5 minutes, then killed the process from task manager.
The same happens if I click on "download" button, the downloads-related screen doesn't appear, just the application goes unresponsive. Also I have Windows device recovery tool installed for the drivers as recommended by the Getting started section.
What could cause this problem?
Thank you in advance.

Paderman, 2022-04-08:
Same problem without any solution ☹️

gus33000, 2024-10-12:
Please refile this issue and attach the extremely important WPinternals.log file and do not clear our the issue template (please fill it properly). This kind of bug report is otherwise not actionable for us if you refuse to fill in the bug reporting template and provide no log file. Thank you.

---

Bug reporting template:

[image]

__Please do not ignore it!__

closed 2024-10-12 by gus33000