ReneLergner/WPinternals #36: can't unlock bootloader on a Nokia Lumia 635

lostgoldplayer, 2022-01-15:


JustNewHere, 2022-08-26:
Follow this guide: https://mega.nz/folder/82YgQabJ#in_YEkxi0vCj_DmBfms5ig/file/B25WHBQY and use this donor FFU: https://mega.nz/folder/82YgQabJ#in_YEkxi0vCj_DmBfms5ig/file/YjxG1LaR

breadbrowser, 2022-08-27:
still didn't work

gus33000, 2024-10-12:
Please refile this issue and attach the extremely important WPinternals.log file and do not clear our the issue template (please fill it properly). This kind of bug report is otherwise not actionable for us if you refuse to fill in the bug reporting template and provide no log file. Thank you.

---

Bug reporting template:

[image]

__Please do not ignore it!__

closed 2024-10-12 by gus33000