ReneLergner/WPinternals #40: HP Elite X3 Windows 10 Mobile Support

SDAdham, 2022-04-01:
Hello everyone,

I am new here, I have HP Elite X3, and done extensive search on how to unlock it and install either W11 or Android (prefer W11 first)

I messed up so much till I get myself locked out with reset protection, here are articles I went through:

- WPInternals doesn't detect my device
- https://gus33000.me/2019/01/05/secureboot-flaw-for-all-wp-devices-literally/ not helpful, a lot of information is missing
- I download qualcom drivers from here https://usb-file.com/download-install-hp-elite-x3-usb-driver/ and here https://android-file.com/download-qualcomm-usb-driver
- Went through the doc here https://woa-project.github.io/LumiaWOA/guides/flashffu/ but i need step 1, your tool to work :D
- I downloaded and installed emergency drivers here https://forum.xda-developers.com/t/emergencydownloaddriver-msi-for-wpinternals-download-link.4398695/
- I downloaded the inf file from here and installed >>  https://raw.githubusercontent.com/kiddlu/windows-android-driver/master/driver/qcom/qcser.inf 

any help would be appreciated, I can't get past the phone to be seen by WP Internals where I can unlock the bootloader

mzayarnyuk, 2022-05-08:
on mi 4 lte for example you can use vol+ and pwr button appear bsd menu;
with vol- select 9 mass storage device; and pwr to select;
now you don't need any tools you can backup full disk or partitions with comodo or acrois or similar raw tools;
and can change registry; with changes file you should be carefull;
better use ffu official tool to custom roms or iutool to add drivers or system cabs packages, etc

mzayarnyuk, 2022-05-08:
if you will have success with old drivers and new arm OS please post here details

SDAdham, 2022-05-09:
@mzayarnyuk i'm not sure if mi 4 lte is hp elite x3, 2 totally different phones afaik? I don't have bsd menu on vol+ and power.

mzayarnyuk, 2022-05-10:
yes, forgot is on new CPU and go with windows onboard;
if you have android version of phone you can try reboot to mass storage and backup;
(for mi4 it was; after enable developer mode and debug via usb and connect like mtp;  adb reboot edl)
hp looks like added special option oem bootloader unlock;
__
pls check this one too:
https://forum.xda-developers.com/t/guide-how-to-dump-and-write-back-the-storage-on-most-of-qualcomm-devices.3949588/
and 
https://www.mobilewithdrivers.com/unlock-bootloader/hp-elite-x3-64gb-4gb-ram
but be cerefull; and be carefull with wpinternals (use it only for backup or flash not for unlock etc; maybe for get root, but don't tried yet...)

closed 2024-10-12 by gus33000