ReneLergner/WPinternals #41: PIN code must be disabled before unlocking bootloader

unilock, 2022-04-27:
I found this out the hard way after flashing my Lumia 950 XL, installing all updates, and setting it up twice over only to have the bootloader unlock fail each time due to E:\ using an "unrecognized file system". (note that WPInternals was looking for "winload.efi" on this volume at the time that it failed)

After attempting to "Restore [the] bootloader" through WPInternals after the second unlock attempt, my Lumia booted to a Windows 7 WBM-esque menu that told me I needed to insert a key device or somesuch in order to unlock BitLocker. So I'm guessing E:\ is encrypted when a PIN code is set up.

Basically, if a PIN code is enabled within Windows 10 Mobile, the bootloader unlock will fail due to being unable to access encrypted files on the device while it's in Mass Storage mode.

(I'm asking that this be added as a warning to WPInternals itself and/or Lumia WoA Project's "[Unlock the Bootloader](https://woa-project.github.io/LumiaWOA/guides/unlock/)" page. I'd do it myself but I'm unfamiliar with the formatting standards of these projects ;) )

unilock, 2022-04-27:
Revision: It's possible that the relevant partition being encrypted was due to enabling "Device encryption" within Windows 10 Mobile.

I do recall, however, that every time I attempted to enable device encryption, I was able to turn the relevant switch to its "on" position, but exiting and re-entering the "Device encryption" menu would reset the switch to its "off" position.

Maybe device encryption just takes a while to set up, and the setting reports itself as "off" until the encryption process is complete? That would be confusing, but I'm guessing that *is* the answer, as it seems like something Windows would do. Sigh.

(I assume this still mandates a warning :) )

gus33000, 2024-10-12:
Please refile this issue and attach the extremely important WPinternals.log file and do not clear our the issue template (please fill it properly). This kind of bug report is otherwise not actionable for us if you refuse to fill in the bug reporting template and provide no log file. Thank you.

---

Bug reporting template:

[image]

__Please do not ignore it!__

closed 2024-10-12 by gus33000

unilock, 2024-10-13:
I am fairly certain the issue template did not exist at the time I filed this issue. Either way, I'm no longer in the possession of the hardware required to reproduce this issue.

gus33000, 2024-10-15:
[image]

oh it did but dont worry about it then, we were cleaning up issues the other day and went through all of them and posted 1:1 the same message