ReneLergner/WPinternals #42: microsoft lumia 640

mzayarnyuk, 2022-04-30:
### Description

I had bricked 640 phone; 
was able to run flash mode with lumia nsu app and power + vol down;
tried flesh with official ffu; completed ok, but stll bricked (amybe hardware errors, etc);
decided to try bootloader unlock (think is not work so will try)
used official ffu and ede;
used RM1152_1078.0037.10586.13080.15864.034DCD_retail_prod_signed.ffu
for profiling;
after onlystart and reboot it ask to reboot manual I used and nothing;
wait a lot still noting;
reboot - still same progress 0%;

now phone in boot loop;
device manager detect it like input USB device....
can you advice some workable revert option or not ?

thx

### Expected behavior

if check logs looks like your app known that it's lumia,
maybe exists some way to rewrite or revert bootloader ?
maybe exists some alternative ? 

can add full log but there exists emei, so shared only bottom part of log.

### WPinternals Version

2.9.7894.11970

### Windows Version

10 - 1703

### Phone model

640 ds

### Relevant Assets

2022-04-30 08:39:30.309: JTAG: Disabled
2022-04-30 08:39:30.310: Bootloader: Lumia Bootloader Spec B
2022-04-30 08:39:30.310: ProductCode: 059X1P2
2022-04-30 08:39:30.310: ProductType: RM-1077
2022-04-30 08:42:31.436: Bootloader was not unlocked. First unlock bootloader before you try to switch to Mass Storage Mode.
2022-04-30 08:51:17.998: Effective Bootloader Security Status: True
2022-04-30 09:29:50.051: Adding FFU to repository: C:\ProgramData\Microsoft\Packages\Products\RM-1077\RM1077_02177.00000.15184.36002_RETAIL_prod_signed_1017_025FF3_000-RU_MV.ffu
2022-04-30 09:29:50.052: Platform ID: Nokia.MSM8226.P6204
2022-04-30 09:29:50.052: Firmware version: 02177.00000.15184.36000
2022-04-30 09:29:50.052: OS version: 6.3.9651.2
2022-04-30 09:29:50.065: Adding emergency files to repository: C:\ProgramData\Microsoft\Packages\Products\RM-1077\MPRG8x26_fh.ede
2022-04-30 09:29:50.065: Type: RM-1077
2022-04-30 09:29:50.133: Adding FFU to repository: E:\downloads\RM1152_1078.0037.10586.13080.15864.034DCD_retail_prod_signed.ffu
2022-04-30 09:29:50.134: Platform ID: Microsoft.MSM8909.P6237
2022-04-30 09:29:50.134: Firmware version: 01078.00037.16082.41000
2022-04-30 09:29:50.134: OS version: 10.0.10586.107
2022-04-30 09:29:50.135: Unlock Bootloader
2022-04-30 09:29:50.135: Processing resources:
2022-04-30 09:29:50.135: Profile FFU: C:\ProgramData\Microsoft\Packages\Products\RM-1077\RM1077_02177.00000.15184.36002_RETAIL_prod_signed_1017_025FF3_000-RU_MV.ffu
2022-04-30 09:29:50.135: EDE file: C:\ProgramData\Microsoft\Packages\Products\RM-1077\MPRG8x26_fh.ede
2022-04-30 09:29:50.135: Donor-FFU with supported OS version: E:\downloads\RM1152_1078.0037.10586.13080.15864.034DCD_retail_prod_signed.ffu
2022-04-30 09:29:50.641: Assembling data for unlock
2022-04-30 09:29:50.642: Assembling data for unlock
2022-04-30 09:29:50.675: Attempt patch: SecureBootHack-V2-EFIESP
2022-04-30 09:29:50.678: Pattern: 0, 0
2022-04-30 09:29:50.678: Pattern: 1, 0
2022-04-30 09:29:50.678: Pattern: 2, 0
2022-04-30 09:29:50.678: Pattern: 3, 0
2022-04-30 09:29:50.678: Pattern: 4, 0
[... 620 lines cut ...]
2022-04-30 09:49:22.394: Mode: Qualcomm Emergency Charging F006
2022-04-30 09:49:30.701: Found device on interface: 4d1e55b2-f16f-11cf-88cb-001111000030
2022-04-30 09:49:30.701: Device path: \\?\HID#VID_05C6&PID_F006#6&12c1f147&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
2022-04-30 09:49:30.701: Connected device: Lumia
2022-04-30 09:49:30.701: Mode: Qualcomm Emergency Charging F006
2022-04-30 09:49:38.951: Found device on interface: 4d1e55b2-f16f-11cf-88cb-001111000030
2022-04-30 09:49:38.951: Device path: \\?\HID#VID_05C6&PID_F006#6&12c1f147&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
2022-04-30 09:49:38.951: Connected device: Lumia
2022-04-30 09:49:38.951: Mode: Qualcomm Emergency Charging F006
2022-04-30 09:49:47.248: Found device on interface: 4d1e55b2-f16f-11cf-88cb-001111000030
2022-04-30 09:49:47.248: Device path: \\?\HID#VID_05C6&PID_F006#6&12c1f147&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
2022-04-30 09:49:47.248: Connected device: Lumia
2022-04-30 09:49:47.248: Mode: Qualcomm Emergency Charging F006
2022-04-30 09:49:55.520: Found device on interface: 4d1e55b2-f16f-11cf-88cb-001111000030
2022-04-30 09:49:55.520: Device path: \\?\HID#VID_05C6&PID_F006#6&12c1f147&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
2022-04-30 09:49:55.520: Connected device: Lumia
2022-04-30 09:49:55.520: Mode: Qualcomm Emergency Charging F006
2022-04-30 09:50:03.820: Found device on interface: 4d1e55b2-f16f-11cf-88cb-001111000030
2022-04-30 09:50:03.820: Device path: \\?\HID#VID_05C6&PID_F006#6&12c1f147&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
2022-04-30 09:50:03.820: Connected device: Lumia
2022-04-30 09:50:03.820: Mode: Qualcomm Emergency Charging F006
2022-04-30 09:50:12.106: Found device on interface: 4d1e55b2-f16f-11cf-88cb-001111000030
2022-04-30 09:50:12.106: Device path: \\?\HID#VID_05C6&PID_F006#6&12c1f147&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
2022-04-30 09:50:12.106: Connected device: Lumia
2022-04-30 09:50:12.106: Mode: Qualcomm Emergency Charging F006

nexus-arch, 2022-08-11:
try to get into flash mode and restore bootloader

mzayarnyuk, 2023-01-24:
thx will try that options;  problem is bootloop and device is not detected and not go in lumia flash mode;
checked other comments maybe problem is I have used pin pass - maybe bitlocker was enabled in that way and it's cause problems; or I choose wrong rom for unlock....

gus33000, 2024-10-12:
Your phone battery is dead, thats what F006 is about

closed 2024-10-12 by gus33000