ReneLergner/WPinternals #44: Error while Unlock bootloader "Error 0x1106: Security header validation failed"

nexus-arch, 2022-08-02:
### Description

When i try to unlock bootloader i gets this error "Error 0x1106: Security header validation failed"
[image]

### Steps To Reproduce

1. Connect lumia 950 to computer.
2. Click Unlock bootloader
3. Click OK
4. After this select file (ffu, edp)
5. And i gets  "Error 0x1106: Security header validation failed"

### Expected behavior

Get a screen with manual reboot

### WPinternals Version

v2.9

### Windows Version

Windows 10 21H1 19043.1826

### Phone model

Lumia 950

### Relevant Assets

[WPInternals.log](https://github.com/ReneLergner/WPinternals/files/9242378/WPInternals.log)

Jaifroid, 2022-09-15:
Hey @nexus-arch, did you check this issue: https://github.com/ReneLergner/WPinternals/issues/30 ? I think it's the same issue, and has a solution that worked for me.

nexus-arch, 2022-09-15:
Nope, i have still same problem