ReneLergner/WPinternals #48: Failed to write to pipe

ShibeGG1, 2023-01-08:
### Description

when trying to unlock bootloader on both lumia 735 and lumia 930 it says failed to write to pipe when its scanning for flashing profile. not sure if this helps but it says: Failed to write to pipe: 4E4FB58464645

### Steps To Reproduce

1. Click download on the side
2. Download the FFU for your phone
3. It downloads the FFU, emergency programmer and donor FFU (950)
4. Once downloaded, click Unlock bootloader
5. Enter in your FFU's and the emergency programmer
6. Click unlock
7. wait a few seconds and it will come up with failed to write to pipe.

### Expected behavior

I expected it to unlock normally and boot into mass storage mode.

### WPinternals Version

v2.9

### Windows Version

Edition	Windows 11 Home Version	22H2 Installed on	‎04/‎01/‎2023 OS build	22621.963 Experience	Windows Feature Experience Pack 1000.22638.1000.0

### Phone model

Lumia 735 (RM-1038) and also Lumia 930 (RM-1045)

### Relevant Assets

[WPInternals.log](https://github.com/ReneLergner/WPinternals/files/10368535/WPInternals.log)

Daniel224455, 2023-02-25:
I have this exact same error here, also running windows 11 might be an issue. im not really sure

Daniel224455, 2023-03-13:
Tested with Windows 10 and it got further, try installing 10 instead of 11

Pinguin2001, 2023-03-22:
Do you have an sd card inserted?

PiQuK, 2023-08-06:
The problem appears while writing part 3 of. After that, there's red text "Not for resale" on left bottom Lumia's screen (under the Windows logo). Tested with Windows 10 x64.

gus33000, 2024-10-12:
Please try using WPinternals 3.0 preview 3, if the issue persists, please file a new issue

closed 2024-10-12 by gus33000