ReneLergner/WPinternals #49: Not working for lumia 735?

StarNumber12046, 2023-04-09:
### Description

The FFU-image you selected for profiling does not have a supported OS-version. 

### Steps To Reproduce

1. Download files for RM-1041 (prod id 059W946)
2. Try to unlock bootloader
3. The error

### Expected behavior

It should allow me to unlock the bootloader

### WPinternals Version

v2.9.2

### Windows Version

windows 11 22h2 10.0.22621.1485

### Phone model

Nokia Lumia 735

### Relevant Assets

2023-04-09 15:36:24.512: Windows Phone Internals version 2.9.7894.11970
2023-04-09 15:36:24.554: Copyright Heathcliff74 / wpinternals.net
2023-04-09 15:36:28.513: Found device on interface: 08324f9c-b621-435c-859b-ae4652481b7c
2023-04-09 15:36:28.513: Device path: \\?\usb#vid_0421&pid_06fc&mi_03#6&1b948965&0&0003#{08324f9c-b621-435c-859b-ae4652481b7c}
2023-04-09 15:36:28.514: Connected device: Lumia
2023-04-09 15:36:28.514: Mode: Normal
2023-04-09 15:36:28.657: Operator: 000-IT
2023-04-09 15:36:28.660: Manufacturer Model Name: RM-1038_1012
2023-04-09 15:36:28.747: Product Code: 059W558
2023-04-09 15:36:28.749: Firmware: 02177.00000.15184.36004
2023-04-09 15:36:28.761: HWID: 1000
2023-04-09 15:36:28.784: IMEI: 354265063491840
2023-04-09 15:36:28.793: Public ID: A1 FA 2A 0E 93 D8 84 83 A6 DD D9 52 19 D1 2A A5 AC A7 55 D3
2023-04-09 15:36:28.798: Bluetooth MAC: D0 92 9E 7E 7C 93
2023-04-09 15:36:28.816: WLAN MAC 1: D0 92 9E 80 CC 9C
2023-04-09 15:36:28.835: WLAN MAC 2: D0 92 9E 80 CC 9D
2023-04-09 15:36:28.853: WLAN MAC 3: D0 92 9E 80 CC 9E
2023-04-09 15:36:28.871: WLAN MAC 4: D0 92 9E 80 CC 9F
2023-04-09 15:36:28.889: Bootloader Security: Enabled
2023-04-09 15:36:28.900: Simlock: Unlocked
2023-04-09 15:36:29.159: Boot policy: Unsecure
2023-04-09 15:36:29.159: DB: SecureProduction
2023-04-09 15:36:29.160: DBX: SecureProduction
2023-04-09 15:36:29.160: KEK: SecureProduction
2023-04-09 15:36:29.160: PK: SecureProduction
2023-04-09 15:36:36.072: Rebooting phone to Flash mode
2023-04-09 15:36:36.075: Rebooting phone to Flash mode...
2023-04-09 15:36:39.468: Start unlock. Phone needs to switch to Flash-mode
2023-04-09 15:36:39.468: In order to start unlocking the bootloader, the phone needs to be switched to Flash-mode.
2023-04-09 15:36:41.882: Switching to Flash-mode
2023-04-09 15:36:41.883: Rebooting phone to Flash mode
2023-04-09 15:36:41.884: Rebooting phone to Flash mode...
2023-04-09 15:36:50.042: Lumia disconnected
2023-04-09 15:36:56.518: Flash App Type: FlashApp
2023-04-09 15:36:56.522: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2023-04-09 15:36:56.522: Device path: \\?\USB#VID_0421&PID_0714#5&3b099d30&0&1#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2023-04-09 15:36:56.522: Connected device: Lumia
2023-04-09 15:36:56.522: Mode: Flash
2023-04-09 15:36:56.766: Phone type: RM-1038
2023-04-09 15:36:56.766: Product code: 059W558
2023-04-09 15:36:56.766: Root key hash: B68498287D74BB4247C4F8D0C3B19C1FCBCE9AB4D770F84E795B53C594F4627C
2023-04-09 15:36:56.766: Firmware version: 02177.00000.15184.36000
[... 178 lines cut ...]
2023-04-09 17:15:04.118: ProductCode: 059W558
2023-04-09 17:15:04.118: ProductType: RM-1038
2023-04-09 17:15:30.605: Effective Bootloader Security Status: True
2023-04-09 17:15:34.468: Bootloader was not unlocked. First unlock bootloader before you try to switch to Mass Storage Mode.
2023-04-09 17:16:05.951: Error: Bad FFU file
2023-04-09 17:16:05.951: Bad FFU file: C:\ProgramData\WPInternals\Repository\RM-1041\RM1041_fh.edp.
Format of the executable (.exe) or library (.dll) is invalid.
2023-04-09 17:16:05.979:    at WPinternals.FFU..ctor(String Path)
   at WPinternals.WPinternalsConfig.AddFfuToRepository(String FFUPath)
2023-04-09 17:16:05.979:     Error: Format of the executable (.exe) or library (.dll) is invalid.
2023-04-09 17:16:05.979:        at WPinternals.FFU..ctor(String Path)
2023-04-09 17:16:05.980: Adding emergency files to repository: C:\ProgramData\WPInternals\Repository\RM-1041\MPRG8x26_fh.ede
2023-04-09 17:16:05.980: Type: RM-1038
2023-04-09 17:16:06.089: Unlock Bootloader
2023-04-09 17:16:06.089: Processing resources:
2023-04-09 17:16:06.089: Profile FFU: C:\ProgramData\WPInternals\Repository\RM-1041\RM1041_fh.edp
2023-04-09 17:16:06.089: EDE file: C:\ProgramData\WPInternals\Repository\RM-1041\MPRG8x26_fh.ede
2023-04-09 17:16:06.115: Error: Bad FFU file
2023-04-09 17:16:06.116: Bad FFU file: C:\ProgramData\WPInternals\Repository\RM-1041\RM1041_fh.edp.
Format of the executable (.exe) or library (.dll) is invalid.
2023-04-09 17:16:06.117:    at WPinternals.FFU..ctor(String Path)
   at WPinternals.LumiaUnlockBootloaderViewModel.LumiaUnlockUEFI(PhoneNotifierViewModel Notifier, String ProfileFFUPath, String EDEPath, String SupportedFFUPath, SetWorkingStatus SetWorkingStatus, UpdateWorkingStatus UpdateWorkingStatus, ExitSuccess ExitSuccess, ExitFailure ExitFailure, Boolean ExperimentalSpecBEFIESPUnlock, Boolean ExperimentalSpecAEFIESPUnlock, Boolean ReUnlockDevice)
2023-04-09 17:16:06.117:     Error: Format of the executable (.exe) or library (.dll) is invalid.
2023-04-09 17:16:06.117:        at WPinternals.FFU..ctor(String Path)
2023-04-09 17:16:06.118: Bad FFU file

StarNumber12046, 2023-04-09:
[image]

gus33000, 2023-04-10:
This is working as expected. You need a donor ffu with a supported os version (does not have to be for the same device either)

closed 2023-04-10 by gus33000