ReneLergner/WPinternals #50: failed to write to pipe 4e4f4b5846

xplanthris, 2023-05-27:
### Description

The bootloader fails to unlock every time on two laptops with the same error of failed to write to pipe 4r4f4b584652004

### Steps To Reproduce

Go to unlock the bootloader
add all the required files
start
wait
get the error after it attempts to flash the unlocked bootloader (part 1)

### Expected behavior

The bootloader should be unlocked normally.

### WPinternals Version

v2.9.2

### Windows Version

Windows 11 22H2 22621.1702

### Phone model

Lumia 550 

### Relevant Assets

2023-05-27 12:16:49.215: Windows Phone Internals version 2.9.7894.11970
2023-05-27 12:16:49.219: Copyright Heathcliff74 / wpinternals.net
2023-05-27 12:16:49.381: Flash App Type: FlashApp
2023-05-27 12:16:49.388: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2023-05-27 12:16:49.388: Device path: \\?\usb#vid_045e&pid_0a02#5&310c583c&0&1#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2023-05-27 12:16:49.388: Connected device: Lumia
2023-05-27 12:16:49.388: Mode: Flash
2023-05-27 12:16:49.472: Security flags: 0x00004BFF
2023-05-27 12:16:49.788: Platform Name: Microsoft.MSM8909.P6236.2.0
2023-05-27 12:16:49.808: Public ID: F0 BF 21 02 92 C6 04 C5 7F 4D 19 56 9A 22 46 20 11 25 DD 9C
2023-05-27 12:16:49.826: Root Key Hash: 56 BD C3 74 62 26 E0 80 14 6D 4D 30 C8 C8 43 26 57 A5 28 2F 05 0B 19 94 CF E8 49 71 C5 86 FA B9
2023-05-27 12:16:49.826: Platform Secure Boot Status: True
2023-05-27 12:16:49.826: Uefi Secure Boot Status: True
2023-05-27 12:16:49.827: Effective Secure Boot Status: True
2023-05-27 12:16:49.827: Bootloader Security Qfuse Status: True
2023-05-27 12:16:49.827: Bootloader Security Authentication Status: False
2023-05-27 12:16:49.827: Bootloader Security Rdc Status: False
2023-05-27 12:16:49.827: Effective Bootloader Security Status: True
2023-05-27 12:16:49.827: Native Debug Status: False
2023-05-27 12:16:49.980: Charging status: 9% - 1867 mA (charging)
2023-05-27 12:16:50.219: Phone type: RM-1127
2023-05-27 12:16:50.219: Product code: 059X5V9
2023-05-27 12:16:50.219: Root key hash: 56BDC3746226E080146D4D30C8C8432657A5282F050B1994CFE84971C586FAB9
2023-05-27 12:16:50.219: Firmware version: 01078.00042.16352.50000
2023-05-27 12:16:50.219: IMEI: 355132070314093
2023-05-27 12:16:50.219: Flash app: 2.75
2023-05-27 12:16:50.219: Flash protocol: 2.41
2023-05-27 12:16:50.219: SecureBoot: Enabled (Platform Secure Boot: Enabled, UEFI Secure Boot: Enabled)
2023-05-27 12:16:50.219: Flash app security: Enabled (FFU security: Enabled, RDC: Not found, Authenticated: False)
2023-05-27 12:16:50.219: JTAG: Disabled
2023-05-27 12:16:50.219: Bootloader: Lumia Bootloader Spec B
2023-05-27 12:16:50.219: ProductCode: 059X5V9
2023-05-27 12:16:50.219: ProductType: RM-1127
2023-05-27 12:17:03.388: FFU not added, because it was already present in the repository.
2023-05-27 12:17:03.388: Emergency files not added, because they were already present in the repository.
2023-05-27 12:17:03.388: Unlock Bootloader
2023-05-27 12:17:03.388: Processing resources:
2023-05-27 12:17:03.388: Profile FFU: C:\ProgramData\WPInternals\Repository\RM-1127\RM1127_1078.0042.10586.13333.15316.03531B_retail_prod_signed.ffu
2023-05-27 12:17:03.388: EDE file: C:\ProgramData\WPInternals\Repository\RM-1127\MPRG8909_fh.ede
2023-05-27 12:17:03.665: Assembling data for unlock
[... 186 lines cut ...]
2023-05-27 12:23:33.679: Connected device: Lumia
2023-05-27 12:23:33.679: Mode: Bootloader
2023-05-27 12:23:39.664: Lumia disconnected
2023-05-27 12:24:22.867: Flash App Type: BootManager
2023-05-27 12:24:22.867: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2023-05-27 12:24:22.867: Device path: \\?\USB#VID_045E&PID_0A02#5&310c583c&0&1#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2023-05-27 12:24:22.867: Connected device: Lumia
2023-05-27 12:24:22.867: Mode: Bootloader
2023-05-27 12:24:28.860: Lumia disconnected
2023-05-27 12:25:12.027: Flash App Type: BootManager
2023-05-27 12:25:12.027: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2023-05-27 12:25:12.027: Device path: \\?\USB#VID_045E&PID_0A02#5&310c583c&0&1#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2023-05-27 12:25:12.027: Connected device: Lumia
2023-05-27 12:25:12.027: Mode: Bootloader
2023-05-27 12:25:18.011: Lumia disconnected
2023-05-27 12:25:59.535: Found device on interface: 4d1e55b2-f16f-11cf-88cb-001111000030
2023-05-27 12:25:59.535: Device path: \\?\HID#VID_05C6&PID_F006#6&14976d43&1&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
2023-05-27 12:25:59.535: Connected device: Lumia
2023-05-27 12:25:59.535: Mode: Qualcomm Emergency Charging F006
2023-05-27 12:26:06.821: Flash App Type: BootManager
2023-05-27 12:26:06.821: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2023-05-27 12:26:06.821: Device path: \\?\USB#VID_045E&PID_0A02#5&310c583c&0&1#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2023-05-27 12:26:06.821: Connected device: Lumia
2023-05-27 12:26:06.821: Mode: Bootloader
2023-05-27 12:26:12.803: Lumia disconnected

gus33000, 2023-05-27:
What's the CPU vendor out of curiosity? AMD? Intel? What generation if possible? Trying to gather what controllers are commonly seeing this behavior, i have some ideas to tackle this as well but need to know the issue at scale

xplanthris, 2023-05-27:
Intel

xplanthris, 2023-05-27:
However an AMD laptop has the same problems

gus33000, 2023-05-27:
and the generation of both?

xplanthris, 2023-05-27:
5950X for the AMD and Core I7 10510U

gus33000, 2023-05-27:
USB-C or USB-A controller? What's the max speed supported on both controllers?

xplanthris, 2023-05-27:
USB-A, how do I figure out the max speed part

gus33000, 2023-05-27:
See the answer on this page: https://superuser.com/questions/478184/verifying-usb-connection-speed-usb-3-or-usb-2

ZecosMAX, 2023-05-31:
@gus33000 lumia 550 has a micro-usb 2.0 port shouldn't PC port even if it is USB 3.0 roll back to 2.0?

gus33000, 2023-05-31:
> @gus33000 lumia 550 has a micro-usb 2.0 port shouldn't PC port even if it is USB 3.0 roll back to 2.0?

Yes but this isn't why I am asking above questions here. I am trying to get a repro environment going.

xplanthris, 2023-06-02:
i couldnt get the speeds but according to one of the answers there are rates and so 
high for the intel laptop
and full for the amd

xplanthris, 2023-06-02:
also now for other devices ive noticed that wpinternals doesnt acknoledge the device sometimes in flash mode

immortal640, 2024-06-14:
Description
The bootloader fails to unlock every time on two laptops with the same error of failed to write to pipe 4r4f4b584652004

Steps To Reproduce
Go to unlock the bootloader
add all the required files
start
wait
get the error after it attempts to flash the unlocked bootloader (part 1)

Expected behavior
The bootloader should be unlocked normally.

WPinternals Version
v2.9.2

Windows Version
Windows 11 22H2 22621.1702

Phone model
Lumia 640XL non LTE DualSim

Relevant Assets
[WPInternals.log](https://github.com/user-attachments/files/15841919/WPInternals.log)

gus33000, 2024-10-12:
Please try using WPinternals 3.0 preview 3, if the issue persists, please file a new issue

closed 2024-10-12 by gus33000