ReneLergner/WPinternals #54: Bootloop after unlocking bootloader.

Wojteczekmumi, 2023-07-07:
Hi. I was trying to unlock bootloader on my lumia 650. during the unlocking there was no errors but after unlocking there was a microsoft logo. then windows logo appear for a second and there was showing like it was updating. then it was rebooting and after 1 hour its still the same. after restoring back bootloader the phone works fine. any solutions for this??

natlegal, 2023-07-11:
It seems that if your phone is in Windows 10 Mobile, the unlock bootloader option in WPInternals is not recommended. Using Interop Tools is the safest option at the moment.

gus33000, 2024-10-12:
Please refile this issue and attach the extremely important WPinternals.log file and do not clear our the issue template (please fill it properly). This kind of bug report is otherwise not actionable for us if you refuse to fill in the bug reporting template and provide no log file. Thank you.

---

Bug reporting template:

[image]

__Please do not ignore it!__

closed 2024-10-12 by gus33000