ReneLergner/WPinternals #55: Crash when flashing original ROM (.ffu)

natlegal, 2023-07-11:
### Description

The phone is correctly detected by the software, I'm able to pause the boot process and enter flash mode. However, the program closes without further explanation when clicking on the "Flash FFU Image" button (after selecting the FFU corresponding to my device). In an attempt to help identify the cause of the bug, I will provide the log.
About the phone: I'm trying to recover a Lumia 640, RM-1109. It was working as expected and had problems after trying to unlock the bootloader. Now the device starts, but does not complete the boot, showing a sad face on a blue background, and then restarts.
Despite this, the device is still detected by WPInternals at boot time by interrupting the boot process. But I can't reinstall the original system using the flash area, since the program closes at the beginning of the process.

### Steps To Reproduce

Go to flash original FFU
Select appropriate FFU file, downloaded via the download section of the application
Press flash FFU image button
WPInternals hangs and then crashes (sometimes it just hangs, but the log shows that nothing is being done in the background)

### Expected behavior

Crash or hangup after pressing flash FFU imagem button

### WPinternals Version

v2.9.2

### Windows Version

Windows 11 22H2 22621.1928

### Phone model

Lumia 640 (RM-1109)

### Relevant Assets

[WPInternals.log](https://github.com/ReneLergner/WPinternals/files/12017874/WPInternals.log)

natlegal, 2023-07-11:
For those who are stuck with the same problem as me. I've found a solution thanks to @gmirz2005 on XDA forum

The solution is:
(You need to have the FFU and other files necessary for your device, you can download in WDRT or WPInternals)

Try removing your battery, connect it to your computer, wait until the computer successfully installs the QH-USB Dloader driver.
Then, try repairing your phone by doing the following steps:
1. start CMD as administrator
2. type cd C:\Program Files (x86)\Microsoft Care Suite\Windows Device Recovery Tool
or if your pc is 32 bit OS, then type C:\Program Files\Microsoft Care Suite\Windows Device Recovery Tool
3. now type the following thor2 -mode emergency -protocol sahara -hexfile edefile.ede -edfile edpfile.edp (replace edefile and edpfile with the files that are adequate to your model)
4. do it for 5 times
5. disconnect the device
6. insert the battery
7. connect it to wall charger
8. wait 10 minutes, for it to charge
9. connect the device to PC
10. if the device shows the red flash screen, then you are good to go
11. if not, then remove the device from computer, disconnect the battery, reconnect the device to computer, launch WDRT
12. click My Device Is Not Detected
13. click LUMIA
14. if your device has been detected, it will be shown in WDRT (if not then retry the steps from 1 to 7 for several times)
15. click your device
16. wait until the device recovers from Emergency State
17. wait until your device reflashes
18. enjoy!

NOTE: All data is already lost and cannot be recovered as Windows Phones use secure erase, and no data recovery tool will be able to get the files back

SO FINGERS CROSSED TO ONEDRIVE BACKUP YOUR PHOTOS/VIDEOS and ELSE!

natlegal, 2023-07-11:
Remember to NEVER UNLOCK BOOTLOADER USING WPINTERNALS when in Windows 10 Mobile. Use Interop Tools instead.

closed 2023-07-11 by natlegal

natlegal, 2023-07-11:
This procedure also fixes glance issues, in case of need.

thanone2, 2025-02-17:
> For those who are stuck with the same problem as me. I've found a solution thanks to @gmirz2005 on XDA forum
>
> The solution is: (You need to have the FFU and other files necessary for your device, [...]

where do I find my ede and edp files?