ReneLergner/WPinternals #58: Lumia 950 "Failed to unlock bootloader"

marufmoinuddin, 2023-10-21:
### Description

I cannot unlock the bootloader on my Lumia 950 (RM-1104). It goes to ass storage mode with an assigned letter for MainOS (F:) but the app still keeps showing switching to mass storage mode. So I force reboot and then after a few steps the bootloader unlocked failed message appears. 

### Expected behavior

Unlock Bootloader

### WPinternals Version

v2.9.1

### Windows Version

Windows 10 22H2

### Phone model

Lumia 950 (RM - 1104)

### Relevant Assets

[WPInternals.log](https://github.com/ReneLergner/WPinternals/files/13060110/WPInternals.log)

marufmoinuddin, 2024-01-22:
> I have the same issue was this ever resolved. I tried to restore the Phone with the Nokia Recovery tool and it made it worse was that locks the phone on the Protection Recovery mode issue

I had to use a clean pc where wpinternals never opened. Also I updated Windows 10 Mobile to its latest build. And the phone got unlocked

gus33000, 2024-10-12:
Please try using WPinternals 3.0 preview 3, if the issue persists, please file a new issue

closed 2024-10-12 by gus33000