ReneLergner/WPinternals #63: supposedly supported OS version (8.10.12393.890) not supported?

frgazor, 2024-01-30:
I'm trying to unlock lumia 520 (RM-914) with OS version 8.10.12393.890 that is supposed to be supported by WPI.  

I have tried  to use ffus downloaded by both WPI and by WPRT (they are identical).  I have also succesfully flashed this ffu on my phone so both phone and ffu OS versions are identical.

But no matter what I do and which donor ffu I choose (RM-1085 seems to be popular) I can't get past this error:
[image]

gus33000, 2024-02-02:
Hi, the supported OS version list is a different thing than picking a donor FFU image. It is possible your RM-1085 ffu image is still not the right supported version (which again, is different from the overall OS support list), not all RM1085 ffus contain an operating system version which will be accepted.

BrownNPC, 2024-06-11:
> not all RM1085 ffus contain an operating system version which will be accepted.

How will I know which one is accepted? Im also having this bug. I downloaded the ffu file through WPInternals 
it's called ```RM914_3058.50000.1425.0001_RETAIL_im_mea3_380_01_446538_prd_signed.ffu```

gus33000, 2024-06-12:
> > not all RM1085 ffus contain an operating system version which will be accepted.
> 
> How will I know which one is accepted? Im also having this bug. I downloaded the ffu file through WPInternals 
> it's called ```RM914_3058.50000.1425.0001_RETAIL_im_mea3_380_01_446538_prd_signed.ffu```
> 
> 

This is only one of the two ffu files WPinternals will have downloaded for you. When you click "download all" WPinternals first downloads your device ffu file (this is the file you quoted above) and only once done, download the supported rm1085 one. This one is guaranteed to work. If you closed the application before this started or did not use "download all" you will have to do this procedure again. Your device own ffu is never the donor ffu. The donor ffu file is a file that can be for a totally different model, it doesn't matter and must be specified in the last field for the donor ffu selection. The first field is for the device own ffu. This must be your device ffu.

gus33000, 2024-06-12:
Closing this issue. Still monitoring it because it is not a bug and functionality works as it should. Ffu files selected by the end user do indeed not contain a supported os version for unlock and the user is highly confused about what to select in the very different donor ffu field, and even swaps both. The user may be confused about donor and device ffus being two widely different things and that donor ffus do not have to be for the same model. Moreover, the user may also be confused as to the selection of the device ffu that must really be their device ffu. It is therefore a misunderstanding of what to select on the user end. I remain available for assistance if the user still didn't figure it out but everything should be explained above.

closed 2024-06-12 by gus33000