ReneLergner/WPinternals #65: No partitions on a Lumia 640 - is it recoverable?

djstevenson123, 2024-02-02:
Here is a question for those who know a lot more about the internal architecture of a Lumia phone than I do, in particular a 640XL but I imagine the same principle will apply to others.

The short summary of the problem I have is that I tried unlocking a Windows phone 10 Lumia 640 with a view to trying out custom images, and it didn't go well.  I can bore people with the longer story if asked, but I have ended up with no partitions on the drive, so nothing happens when I turn on the power, even with the volume down key pressed.  I have tried setting up a sdcard with various boot managers or burned iso's with various OS's and even a copy of what was on the internal drive, with no better result.  Basically, I can't work out a method of bootstrapping it sufficiently to get a system going whereby I can restore the partition table and restore the backup using wpinternals.

It's not the battery - I have fully charged spares. charged from another phone (which I did manage to unlock successfully but haven't yet tried anything risky on it).

The question is - if the phone has no partitions (there were 29!) is there any way of getting enough of it going to be able to see the drive, in which case I can use linux on the host computer to recreate the partition table and 'dd' the data back onto it.

My feeling is that I need to get it to boot from a sdcard with a suitable system on it (don't mind if it is windows, linux or whatever), but although there is advice 'out there' that this is possible I haven't managed to do it.

Bit of a long shot I know, and this may not be relevant here, but if anyone is interested in helping to solve this conundrum I would appreciate it.

gus33000, 2024-02-02:
You want to get the phone recognized onto a computer as QCOMUSB_BLK (or Qualcomm Download Mode 9008). Considering the symptoms, you should already be in it. From there, you can flash an emergency package using Windows Device Recovery Tool (My device is not detected -> Lumia -> pick your rm number) or manually using thor2.

Be aware that specific models of the Lumia 640 may not have such package available or may have mismatched files due to a screwup that happened when microsoft uploaded them years ago and will not be fixed by them at this point.

Be also aware that depending on what you have done, you may have put enough dirty data into vital partitions for, even if you reflash using said emergency package, to not be able to reflash a FFU image once you're back into Nokia's Flash App.

djstevenson123, 2024-02-02:
Thanks for that.  I will look into your first point.  I have tried the WDRT, selecting the phone as suggested (Lumia 640 LTE Dual Sim), but I get 'operation ended in failure'.  I'll have a go with thor2, however I suspect it won't see anything it can work on.  Maybe getting it into QCOMUSB_BLK might help, when I work out how to do it!

I think the data will still be on the drive, but there are no partition boundaries to tell anything where to look.

I appreciate your time, Gustave.

djstevenson123, 2024-02-02:
Well, I installed the Qualcomm HS-USB QDloader 9008 Driver (63 bit) onto a windows 11 laptop.  There is an entry for it under Ports (Com and LPT, on COM3).  An attempt to list devices in adb found nothing, and all thor2 commands resulted in 'no connection' (or similar).  The WDRT found the correct Lumia model (RM-1075) when I selected the Lumia option, but it still failed after downloading the files.  The Nokia flashing tool produced this:'No device detected, make sure it is in fastboot mode (power + volume down)', which I tried, obviously.

Possibly BTW, it seems to work better when the battery is out.  When installed the Ports entity does not appear in device manager (and no beep when it is connected).  Also doesn't show when the sd card is in.

Is it too far gone, or is there anything else worth trying?  Any thoughts would be welcome and appreciated.

djstevenson123, 2024-02-05:
Hi Gustave; thanks for persevering!  I didn't realise that adb was just for
android - thanks for the lesson.

I ran the procedure described in your lin, although I had already
downloaded and used WDRT and WPInternals.  All went ok and the output from
the thor2 command seemed ok until this:

....
Sending RAW_DATA
RAW_DATA sent
END_IMAGE_TX received
END_IMAGE_TX with error status (triggering RESET)
Sending RESET
RESET sent
Entering state WAIT_RESET_RESP
RESET_RESP received
Status code 18, Error Str: Invalid destination address
Stopped Sahara emergency download (status code = 18)
EMERGENCY FLASH END
Emergency messaging closed successfully
Operation took about 1.00 second.

Unknown error code.

THOR2 1.8.2.18 exited with error code 18 (0x12)

Does that help at all with any diagnosis?

Once again I appreciate your time.

djstevenson123, 2024-02-07:
I have done quite a bit to try and flash something onto the phone, using just about every thor2 command there is! (Well not really - there are hundreds of different iterations).  I have also tried installing different qualcomm drivers.  The only way I can get thor2 to even see the lumia is if I take all the drivers out.  In any other sceanario, including the qualcomm 9008 one, thor2 -mode list_connections shoes nothing.  Without any drivers I get

WinUSB in use.
Connection list START
0.1400:0001:0001        {71de994d-8b7c-43db-a27e-2ae7cd579a0c}   Emergency mode connected
Connection list END

The device shows up in device manager as QHSUSB_BULK (in Universal Serial Bus devices).  That seemed a start, but try as I might, I can't get thor2 to flash the emergency partition.  Most of the attempts report 'Device is not in Lumia UEFI mode'.  I had hoped that

thor2 -mode emergency -hexfile MPRG8x26_fh.ede -edfile RM1075_fh.edp -protocol sahara -skipffuflash

but I get 'END_IMAGE_TX with error status (triggering RESET)' and then 'Status code 4, Error Str: Invalid host protocol version'.  (thor2 appears to work only with sahara protocol).

I suspect the problem is that there are no partitions on the disk; attempts to flash any of them with the appropriate 'bin' file (eg EFIESP.bin) fail.  I think I need a way to get them back manually.  In linux I would use fdisk and import from a partition image backup (which I have) but don't know how to do that, if it is even possible, on a lumia 640.

Lots of searches haven't helped.  If this rings a bell with anyone who has experience of this working I would appreciate hearing about it.  Thanks for reading this - sorry it's a bit long.  BTW I have tried this on two different machines - a reasonably modern one with WIndows 11 and a very old one with Windows 10 (32 bit); they seem to behave quite differently, which doesn't help my understanding of all this much ;-(

djstevenson123, 2024-02-10:
I have come to the conclusion that this phone is not recoverable using these software methods, probably because I deleted the partition table on the phone, and that if I am to recover it it will need the hardware to do it (eg thor2 output: Cannot flash partition image. Write the RDC into the device or use open/RnD HW & SW), jTAG I think it is.  So I am giving up.  I have other lumia 640 phones (mainly for interest/research now) so it's not a problem.  If anyone disagrees with this please let me know.  In the meantime, Thanks Gustave for your input; I learnt from it.

closed 2024-10-12 by gus33000