ReneLergner/WPinternals #67: WP internals hanging and freezing when attempting to unlock device

AttackVector, 2024-03-02:
I have tried dozens of cables and the program hangs when i try unlocking the device or do any action. I really want to unlock my WP8.1 Phone. It is a lumia 925 att

JoveyMcJupi, 2024-04-10:
Same issue here. Can't unlock the device, but mine is a Lumia 640 LTE. Running Canary build of Windows 11, might have to do something with it. Will try Windows 7, or 10.

MackaG, 2024-05-05:
did you solve the issue?

gus33000, 2024-10-12:
Please refile this issue and attach the extremely important WPinternals.log file and do not clear our the issue template (please fill it properly). This kind of bug report is otherwise not actionable for us if you refuse to fill in the bug reporting template and provide no log file. Thank you.

---

Bug reporting template:

[image]

__Please do not ignore it!__

closed 2024-10-12 by gus33000