ReneLergner/WPinternals #69: WPInternals crashing/freezing when pressing "unlock" button(RM-1045, Spec B)

MackaG, 2024-05-05:
### Description

I already had this issue on 640 lte while trying to use EDE file, when using EDP everything was fine, until I bought 930 and its simply crashing

### Steps To Reproduce

Connect lumia 930 in flash mode to WPinternals
Download latest firmware, ede and edp files
Download firmware for RM-1085 with compatibile OS version
Select corresponding files
Click "Unlock"

### Expected behavior

WPInternals freezes and closes

### WPinternals Version

2.9.2 and 2.3

### Windows Version

Windows 11 23h2 and Windows 7

### Phone model

Lumia 930 and 640 LTE

### Relevant Assets

I'm on different PC, but I noticed error like: "Indeks wykracza poza granice tablicy" which means Index exceeds numver of array elements

MackaG, 2024-05-05:
Important part of log:
Adding emergency files to repository: C:\ede.ede
Type: RM-1045
Error: Indeks wykracza poza granice tablicy

gus33000, 2024-05-06:
Hi, can you please attach the log file? Thank you

MackaG, 2024-05-06:
Ok

2024-05-05 23:01:28.708: Windows Phone Internals version 2.3.6584.1869
2024-05-05 23:01:28.710: Copyright Heathcliff74 / wpinternals.net
2024-05-05 23:01:28.919: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2024-05-05 23:01:28.919: Device path: \\?\usb#vid_0421&pid_0714#5&fa9ae3&0&4#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2024-05-05 23:01:28.919: Connected device: Lumia
2024-05-05 23:01:28.919: Mode: Flash
2024-05-05 23:01:29.060: Security flags: 0x00007BFF
2024-05-05 23:01:29.087: Platform Name: Nokia.MSM8974.P6180.2.2
2024-05-05 23:01:29.112: Public ID: 6C F2 C3 70 7E BB 00 9A 33 4F 9E EE AE 7F A9 31 A1 A4 35 23
2024-05-05 23:01:29.143: Root Key Hash: 80 0E EB 50 8F 7B BD F1 2A 19 26 26 21 FD 83 72 97 A3 B0 62 FE 2A 70 78 C0 F3 16 7E 57 F2 12 17
2024-05-05 23:01:29.143: Platform Secure Boot Status: True
2024-05-05 23:01:29.144: Uefi Secure Boot Status: True
2024-05-05 23:01:29.144: Effective Secure Boot Status: True
2024-05-05 23:01:29.145: Bootloader Security Qfuse Status: True
2024-05-05 23:01:29.145: Bootloader Security Authentication Status: False
2024-05-05 23:01:29.147: Bootloader Security Rdc Status: False
2024-05-05 23:01:29.147: Effective Bootloader Security Status: True
2024-05-05 23:01:29.147: Native Debug Status: False
2024-05-05 23:01:29.418: Phone type: RM-1045
2024-05-05 23:01:29.418: Product code: 059W0X6
2024-05-05 23:01:29.418: Root key hash: 800EEB508F7BBDF12A19262621FD837297A3B062FE2A7078C0F3167E57F21217
2024-05-05 23:01:29.418: Firmware version: 02540.00019.15236.45000
2024-05-05 23:01:29.418: IMEI: 353633062332557
2024-05-05 23:01:29.419: Flash app: 2.59
2024-05-05 23:01:29.419: Flash protocol: 2.34
2024-05-05 23:01:29.419: SecureBoot: Enabled
2024-05-05 23:01:29.419: Flash app security: Enabled (FFU security: Enabled, RDC: Not found, Authenticated: False)
2024-05-05 23:01:29.419: JTAG: Disabled
2024-05-05 23:01:29.419: Bootloader: Lumia Bootloader Spec B
2024-05-05 23:01:29.419: ProductCode: 059W0X6
2024-05-05 23:01:29.420: ProductType: RM-1045
2024-05-05 23:01:59.953: Adding FFU to repository: C:\ffu1.ffu
2024-05-05 23:01:59.953: Platform ID: Nokia.MSM8974.P6180
2024-05-05 23:01:59.953: Firmware version: 02540.00019.15236.45000
2024-05-05 23:01:59.953: OS version: 6.3.9651.2
2024-05-05 23:02:00.003: Adding emergency files to repository: C:\edp.edp
2024-05-05 23:02:00.003: Type: RM-1045
2024-05-05 23:02:00.014: Error: Indeks wykracza poza granice tablicy.
2024-05-05 23:02:09.229: Windows Phone Internals version 2.3.6584.1869
2024-05-05 23:02:09.230: Copyright Heathcliff74 / wpinternals.net
2024-05-05 23:02:09.439: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2024-05-05 23:02:09.439: Device path: \\?\usb#vid_0421&pid_0714#5&fa9ae3&0&4#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2024-05-05 23:02:09.439: Connected device: Lumia
2024-05-05 23:02:09.439: Mode: Flash
2024-05-05 23:02:09.557: Security flags: 0x00007BFF
2024-05-05 23:02:09.581: Platform Name: Nokia.MSM8974.P6180.2.2
2024-05-05 23:02:09.605: Public ID: 6C F2 C3 70 7E BB 00 9A 33 4F 9E EE AE 7F A9 31 A1 A4 35 23
2024-05-05 23:02:09.626: Root Key Hash: 80 0E EB 50 8F 7B BD F1 2A 19 26 26 21 FD 83 72 97 A3 B0 62 FE 2A 70 78 C0 F3 16 7E 57 F2 12 17
2024-05-05 23:02:09.627: Platform Secure Boot Status: True
2024-05-05 23:02:09.627: Uefi Secure Boot Status: True
2024-05-05 23:02:09.628: Effective Secure Boot Status: True
2024-05-05 23:02:09.628: Bootloader Security Qfuse Status: True
2024-05-05 23:02:09.628: Bootloader Security Authentication Status: False
2024-05-05 23:02:09.629: Bootloader Security Rdc Status: False
2024-05-05 23:02:09.629: Effective Bootloader Security Status: True
2024-05-05 23:02:09.629: Native Debug Status: False
2024-05-05 23:02:09.920: Phone type: RM-1045
2024-05-05 23:02:09.920: Product code: 059W0X6
2024-05-05 23:02:09.920: Root key hash: 800EEB508F7BBDF12A19262621FD837297A3B062FE2A7078C0F3167E57F21217
2024-05-05 23:02:09.920: Firmware version: 02540.00019.15236.45000
2024-05-05 23:02:09.920: IMEI: 353633062332557
2024-05-05 23:02:09.920: Flash app: 2.59
2024-05-05 23:02:09.920: Flash protocol: 2.34
2024-05-05 23:02:09.920: SecureBoot: Enabled
2024-05-05 23:02:09.920: Flash app security: Enabled (FFU security: Enabled, RDC: Not found, Authenticated: False)
2024-05-05 23:02:09.920: JTAG: Disabled
2024-05-05 23:02:09.921: Bootloader: Lumia Bootloader Spec B
2024-05-05 23:02:09.921: ProductCode: 059W0X6
[... 274 lines cut ...]
2024-05-06 02:10:29.227: Flash protocol: 2.34
2024-05-06 02:10:29.227: SecureBoot: Enabled
2024-05-06 02:10:29.227: Flash app security: Enabled (FFU security: Enabled, RDC: Not found, Authenticated: False)
2024-05-06 02:10:29.227: JTAG: Disabled
2024-05-06 02:11:43.426: FFU not added, because it was already present in the repository.
2024-05-06 02:11:43.428: Adding emergency files to repository: C:\edp.edp
2024-05-06 02:11:43.428: Type: RM-1045
2024-05-06 02:11:43.438: Error: Indeks wykracza poza granice tablicy.
2024-05-06 02:36:33.937: Windows Phone Internals version 2.8.6950.39616
2024-05-06 02:36:33.939: Copyright Heathcliff74 / wpinternals.net
2024-05-06 02:36:34.119: Found device on interface: 08324f9c-b621-435c-859b-ae4652481b7c
2024-05-06 02:36:34.119: Device path: \\?\usb#vid_0421&pid_06fc&mi_03#6&3aa93d4e&0&0003#{08324f9c-b621-435c-859b-ae4652481b7c}
2024-05-06 02:36:34.119: Connected device: Lumia
2024-05-06 02:36:34.119: Mode: Normal
2024-05-06 02:36:34.829: Operator: 000-33
[... cut ...]
2024-05-06 02:36:34.906: Product Code: 059W0X6
2024-05-06 02:36:34.909: Firmware: 02540.00019.15236.45005
2024-05-06 02:36:34.925: IMEI: 353633062332557
2024-05-06 02:36:34.948: Public ID: 6C F2 C3 70 7E BB 00 9A 33 4F 9E EE AE 7F A9 31 A1 A4 35 23
2024-05-06 02:36:34.969: Bluetooth MAC: 10 2F 6B BD 91 12
2024-05-06 02:36:35.015: WLAN MAC: 10 2F 6B BF E5 68
2024-05-06 02:36:35.045: Bootloader Security: Enabled
2024-05-06 02:36:35.082: Simlock: Unlocked
2024-05-06 15:18:36.547: Lumia disconnected

gus33000, 2024-05-06:
It appears you are selecting the ede in the "add ffu to repo" feature. This feature can only add ffu files. Not emergency files. You can only specify an ede file (and not an edp file) in the unlock screen where it asks for files in the field to select an emergency programmer. Be aware you may have multiple feeds. One for your ffu image. One for the ede, and one optionally for an ffu with a supported os version as a donnor.

MackaG, 2024-05-06:
It was thing i were trying to do to make tool work. Thats why I included only "important part" of log. WPInternals were doing same thing, outputing same error on a clean Windows 7 install without doing anything. Only thing I did was connect lumia 930 to PC, enter unlock bootloader section and select files that were needed to unlock bootloader, and it crashed. I can provide you log from that windows 7 PC if you ask

MackaG, 2024-05-06:
Accidentally selected close with comment

MackaG, 2024-05-06:
hello?

MackaG, 2024-05-06:
OK, I figured it out. Lumiafirmware.com's ftp protocol is broken and it inserts fragment of HTML script before actual data. When I removed it with HxD it moved further, but there's another error. Good old failed to write to pipe, and reported exception ".".(?). I dont know but I'm happy that at least it partially works. Also, I can use FFU file as emergency programmer for some reason and it gives me same error.

closed 2024-10-12 by gus33000