ReneLergner/WPinternals #71: Supported OS bug on Lumia 820

sprono, 2024-06-05:
### Description

I am pretty sure my Lumia 820 is of the Vietnam sales region and the firmware image I have downloaded is of a supported OS version (FW 3051.50009.1424.0002 on WP8.1 8.10.12393.890.) I have downloaded both the FFU from LumiaFirmware and the firmware that WPI itself suggested, and even the firmware image downloaded by WDRT (which must've been one of its dependencies.)
 
[image]

[image]

However none of these seem to work-and even when its own suggested firmware is not supported, it never indicated that it was going to download another compatible FFU image, instead it prompted me to select another compatible FFU.

[image]

WPI does say FFUs from other devices are supported for donors, but how do I know which one would work for my 820 since all FFUs look pretty arbitrary and I can't get DISM to work with them-it throws a code-31 error instead, signifying the location where log files are to be written is inaccessible (or so I guessed.)

[image]

Or because the Supported OS List for All Lumias only include W10M versions, and WP8.1 is only supported for root access? I can't wrap my head around things. Assistance much appreciated.

### Steps To Reproduce

1. Plug the 820 in.
2. Reboot into Flash Mode.
3. Select proper FFU, emergency loader folder and SBL3 ROM for mass storage.
4. The seemingly-correct FFUs are considered unsupported by WPI as seen in the above images.

### Expected behavior

As 8.10.12393.890 is on the list of supported OSes, it should've been allowed to proceed to unlock the bootloader, however as evidently shown above it does not let me continue flashing at all.

### WPinternals Version

2.9

### Windows Version

Windows 11 24H2 26227.5000

### Phone model

Lumia 820 (RM824)

### Relevant Assets

[WPInternals.log](https://github.com/user-attachments/files/15585337/WPInternals.log)

gus33000, 2024-06-06:
Hi, the supported os list is the list of supported OS versions you can have on your device with the bootloader unlocked. It doesn't have much to do with what the supported ffu inquiry is about and is separate. If you used WPinternals download feature you should have obtained as well a download for the 950 XL ffu. Otherwise, that's what you should download manually from elsewhere.

aervnu, 2024-06-06:
[image]

OK, the Lumia 950XL 10.0.10586.318 ROM passed the OS check. But quickly after that it spawned an error, saying it has failed to parse the SBL1 partition:

[image]

Any way to resolve this? I've verified that the download is intact and not corrupt.

gus33000, 2024-06-06:
Select the Lumia 950 XL ffu for the supported ffu field. The first field has to be your device ffu. It cannot be anything else

aervnu, 2024-06-06:
The thing is if I choose the 820 FFU as the donor FFU (as in first field) then it'll just refuse to proceed as I've stated. Or do I have to find a Windows 10 device that is of Bootloader Spec A? Since from what I've read the SBL partition has something to do with the bootloader, and the 950XL is a Spec B so I think that might be why it couldn't parse the SBL1 partition despite using a FFU with a supported OS.

Does such a device even exist? I'm not sure whether Windows 10 is a Spec B only OS.

gus33000, 2024-06-09:
> The thing is if I choose the 820 FFU as the donor FFU (as in first field) then it'll just refuse to proceed as I've stated, because 820 doesn't officially support Windows 10. Or do I have to find a Windows 10 device that is of Bootloader Spec A? Since from what I've read the SBL partition has something to do with the bootloader, and the 950XL is a Spec B so I think that might be why it couldn't parse the SBL1 partition despite using a FFU with a supported OS.
> 
> Does such a device even exist? I'm not sure whether Windows 10 is a Spec B only OS.

I think you didn't understand.

Select your Lumia 820 FFU for the device FFU (very first selection item)
And select the Lumia 950 XL FFU for the donor FFU (very last selection item)

gus33000, 2024-06-09:
[image]

gus33000, 2024-06-09:
The first field isn't a "donor ffu" field, that's the last field. The first field is your device FFU

gus33000, 2024-06-12:
Closing this issue. Still monitoring it because it is not a bug and functionality works as it should. Ffu files selected by the end user do indeed not contain a supported os version for unlock and the user is highly confused about what to select in the very different donor ffu field, and even swaps both. The user may be confused about donor and device ffus being two widely different things and that donor ffus do not have to be for the same model. Moreover, the user may also be confused as to the selection of the device ffu that must really be their device ffu. It is therefore a misunderstanding of what to select on the user end. I remain available for assistance if the user still didn't figure it out but everything should be explained above.

closed 2024-06-12 by gus33000