ReneLergner/WPinternals #72: WPInternals 3.0 Preview 1 timebomb

AppleInternal, 2024-06-26:
### Description

please remove timebomb

### Steps To Reproduce

open WPInternals 3.0 Preview 1

### Expected behavior

WPInternals 3.0 Preview 1 will launch normally

### WPinternals Version

v3.0

### Windows Version

Windows 11 10.0.26040.1000

### Phone model

Lumia 950 (RM-1118)

### Relevant Assets

Windows Phone Internals version 3.0.8730.32231
Copyright Heathcliff74
This prerelease version is expired!

gus33000, 2024-07-01:
Works as intended

lightmon233, 2025-12-06:
Can I use stable release to unlock lumia 950, there is no 950xl firmware selection box in that version of wpinternal, and I tried many times but still can't unlock the bootloader