ReneLergner/WPinternals #75: Lumia 435 DS with WP8.1

Kellerossel, 2024-09-11:
### Description

I am trying to use WPinternals for a Lumia 435 Double Sim, but if I download the .ffu for it and try to get it running i get the Error `**Bad GPT**. The GPT read isn't valid. Couldn't find text "EFI PART"`
 I tried downloading the ffu from the official site again, but it didn't work. Is there a way to fix this?

### Expected behavior

I expected it to work on first try.

### WPinternals Version

v2.9

### Windows Version

Windows 10 pro

### Phone model

Lumia 435 DS

Cheikhabbas44, 2024-09-11:
Put yourself on Windows Internal section donload and plug in the phone. You will see the firmware that corresponds to your phone and download it. And now you can use them

gus33000, 2024-09-12:
Please try version 2.9.2 and see if the issue remains, and attach logs with 2.9.2, not 2.9 *thats outdated.

It does however sound like your FFU file is corrupt. But update your app version

Kellerossel, 2024-09-13:
Thank you, I will try the suggestions in the next days and give you an update on how it went

Kellerossel, 2024-09-20:
I downloaded it again and installed it. It is still showing me that it has version 2.9 (and it still doesn't work)

gus33000, 2024-09-22:
Can you please send the log files?

Kellerossel, 2024-10-09:
Where do I find the log files?

gus33000, 2024-10-10:
%localprogramdata%\WPinternals\WPinternals.log

Kellerossel, 2024-10-12:
Right now I'm on vacation, but I'll send it in a week or so. Sorry for causing so much Trouble 😵‍💫

Kellerossel, 2024-10-16:
@gus33000 
```2024-09-10 20:55:47.443: Windows Phone Internals version 2.9.7894.11970
2024-09-10 20:55:47.455: Copyright Heathcliff74 / wpinternals.net
2024-09-10 20:57:47.634: Found device on interface: 08324f9c-b621-435c-859b-ae4652481b7c
2024-09-10 20:57:47.634: Device path: \\?\usb#vid_0421&pid_06fc&mi_03#6&36a4fec4&0&0003#{08324f9c-b621-435c-859b-ae4652481b7c}
2024-09-10 20:57:47.634: Connected device: Lumia
2024-09-10 20:57:47.634: Mode: Normal
2024-09-10 20:57:47.783: Operator: 000-DE
2024-09-10 20:57:47.785: Manufacturer Model Name: RM-1069_1003
2024-09-10 20:57:47.807: Product Code: 059X1W8
2024-09-10 20:57:47.810: Firmware: 02074.00000.15234.28002
2024-09-10 20:57:47.813: HWID: 1000
2024-09-10 20:57:47.829: IMEI: 358609064338441
358609064338458
2024-09-10 20:57:47.837: Public ID: C9 4D 6D 6C 82 76 38 86 AC A3 E6 2C E6 3A 6F EB FD C5 2E 79
2024-09-10 20:57:47.845: Bluetooth MAC: 14 9A 10 4D 02 7A
2024-09-10 20:57:47.872: WLAN MAC 1: 14 9A 10 52 95 1C
2024-09-10 20:57:47.902: WLAN MAC 2: 14 9A 10 52 95 1D
2024-09-10 20:57:47.931: WLAN MAC 3: 14 9A 10 52 95 1E
2024-09-10 20:57:47.961: WLAN MAC 4: 14 9A 10 52 95 1F
2024-09-10 20:57:47.975: Bootloader Security: Enabled
2024-09-10 20:57:47.993: Simlock: Unlocked
2024-09-10 20:57:48.135: Boot policy: Unsecure
2024-09-10 20:57:48.135: DB: SecureProduction
2024-09-10 20:57:48.135: DBX: SecureProduction
2024-09-10 20:57:48.135: KEK: SecureProduction
2024-09-10 20:57:48.135: PK: SecureProduction
2024-09-10 20:58:02.485: Start unlock. Phone needs to switch to Flash-mode
2024-09-10 20:58:02.485: In order to start unlocking the bootloader, the phone needs to be switched to Flash-mode.
2024-09-10 20:58:06.696: Switching to Flash-mode
2024-09-10 20:58:06.703: Rebooting phone to Flash mode
2024-09-10 20:58:06.706: Rebooting phone to Flash mode...
2024-09-10 20:58:12.377: Lumia disconnected
2024-09-10 20:58:17.475: Flash App Type: FlashApp
2024-09-10 20:58:17.484: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
2024-09-10 20:58:17.484: Device path: \\?\USB#VID_0421&PID_0714#5&12582ce9&0&1#{9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc}
2024-09-10 20:58:17.484: Connected device: Lumia
2024-09-10 20:58:17.484: Mode: Flash
2024-09-10 20:58:17.822: Phone type: RM-1069
2024-09-10 20:58:17.822: Product code: 059X1W8
2024-09-10 20:58:17.822: Root key hash: 80F2F707F2EF1F5650E81C07A552E2B911896BB046EB7D33D699F736B29FF74B
2024-09-10 20:58:17.822: Firmware version: 02074.00000.15234.28000
2024-09-10 20:58:17.822: IMEI: 358609064338441
2024-09-10 20:58:17.822: Flash app: 2.59
2024-09-10 20:58:17.822: Flash protocol: 2.34
2024-09-10 20:58:17.822: SecureBoot: Enabled (Platform Secure Boot: Enabled, UEFI Secure Boot: Enabled)
2024-09-10 20:58:17.822: Flash app security: Enabled (FFU security: Enabled, RDC: Not found, Authenticated: False)
2024-09-10 20:58:17.822: JTAG: Disabled
2024-09-10 21:02:38.700: Aborting.
2024-09-10 21:02:38.755: Security flags: 0x00004BFF
2024-09-10 21:02:39.158: Platform Name: Nokia.MSM8210.P6200.1.1
2024-09-10 21:02:39.186: Public ID: C9 4D 6D 6C 82 76 38 86 AC A3 E6 2C E6 3A 6F EB FD C5 2E 79
2024-09-10 21:02:39.214: Root Key Hash: 80 F2 F7 07 F2 EF 1F 56 50 E8 1C 07 A5 52 E2 B9 11 89 6B B0 46 EB 7D 33 D6 99 F7 36 B2 9F F7 4B
2024-09-10 21:02:39.214: Platform Secure Boot Status: True
2024-09-10 21:02:39.214: Uefi Secure Boot Status: True
2024-09-10 21:02:39.214: Effective Secure Boot Status: True
2024-09-10 21:02:39.214: Bootloader Security Qfuse Status: True
2024-09-10 21:02:39.214: Bootloader Security Authentication Status: False
2024-09-10 21:02:39.214: Bootloader Security Rdc Status: False
2024-09-10 21:02:39.214: Effective Bootloader Security Status: True
2024-09-10 21:02:39.214: Native Debug Status: False
2024-09-10 21:02:39.404: Charging status: 80% - 247 mA (charging)
2024-09-10 21:02:39.404: Bootloader: Lumia Bootloader Spec B
2024-09-10 21:02:39.404: ProductCode: 059X1W8
2024-09-10 21:02:39.404: ProductType: RM-1069
2024-09-10 21:02:52.436: Rebooting phone to Normal mode
2024-09-10 21:02:52.436: Rebooting phone to Normal mode...
2024-09-10 21:02:52.498: Lumia disconnected
2024-09-10 21:02:57.388: Flash App Type: BootManager
2024-09-10 21:02:57.388: Found device on interface: 9e3bd5f7-9690-4fcc-8810-3e2650cd6ecc
[... 48 lines cut ...]
2024-09-10 21:49:20.379: Flash app security: Enabled (FFU security: Enabled, RDC: Not found, Authenticated: False)
2024-09-10 21:49:20.379: JTAG: Disabled
2024-09-10 21:49:20.435: Bootloader was not unlocked. First unlock bootloader before you try to switch to Mass Storage Mode.
2024-09-10 21:49:24.621: Security flags: 0x00004BFF
2024-09-10 21:49:25.017: Platform Name: Nokia.MSM8210.P6200.1.1
2024-09-10 21:49:25.060: Public ID: C9 4D 6D 6C 82 76 38 86 AC A3 E6 2C E6 3A 6F EB FD C5 2E 79
2024-09-10 21:49:25.088: Root Key Hash: 80 F2 F7 07 F2 EF 1F 56 50 E8 1C 07 A5 52 E2 B9 11 89 6B B0 46 EB 7D 33 D6 99 F7 36 B2 9F F7 4B
2024-09-10 21:49:25.088: Platform Secure Boot Status: True
2024-09-10 21:49:25.088: Uefi Secure Boot Status: True
2024-09-10 21:49:25.088: Effective Secure Boot Status: True
2024-09-10 21:49:25.088: Bootloader Security Qfuse Status: True
2024-09-10 21:49:25.088: Bootloader Security Authentication Status: False
2024-09-10 21:49:25.088: Bootloader Security Rdc Status: False
2024-09-10 21:49:25.088: Effective Bootloader Security Status: True
2024-09-10 21:49:25.088: Native Debug Status: False
2024-09-10 21:49:25.272: Charging status: 97% - 240 mA (charging)
2024-09-10 21:49:25.388: Bootloader: Lumia Bootloader Spec B
2024-09-10 21:49:25.388: ProductCode: 059X1W8
2024-09-10 21:49:25.388: ProductType: RM-1069
2024-09-10 21:49:37.607: Effective Bootloader Security Status: True
2024-09-10 21:50:00.415: Effective Bootloader Security Status: True
2024-09-10 21:50:10.030: Aborting.
2024-09-10 21:50:26.793: Aborting.
2024-09-10 21:50:34.859: Bootloader was not unlocked. First unlock bootloader before you try to switch to Mass Storage Mode.
```

Kellerossel, 2024-10-28:
Just tell me when you know how to fix :)