ReneLergner/WPinternals #76: "Object reference not set to an instance of an object" (2.9.2) and/or "Bootloader locked" (3.0 Pre2)

EpicLPer, 2024-09-22:
### Description

I've been trying to get my Lumia 950 (RM-1104) back to stock to try out a newly released Android ROM, however I can't get my Lumia back to stock.
As a few days ago the official servers have been turned off I've downloaded (hopefully) the right FFU and emergency files and tried flashing those. I also found an old backup dump that I tried to flash, however the same error "Object reference not set to an instance of an object" comes up.

Using the new 3.0 Pre2 version shows "Bootloader locked", however I'm certain that's not the case as I already ran Windows 11 on this device, further the 3.0 version is constantly waiting for the device to be in Flash mode despite it already being in Flash mode. On 2.9.2 it does recognize the device in Flash mode, however it still throws the reference error.

### Steps To Reproduce

1. Connect device
2. Try flashing FFU or unlocking Bootloader
3. "Object reference not set to an instance of an object" error is the result

### Expected behavior

Functions should work as described :(

### WPinternals Version

2.9.2 and 3.0 Pre2

### Windows Version

Windows 11 22H2

### Phone model

Luma 950 (RM-1104)

gus33000, 2024-09-22:
Can you please attach a clean log file of either versions?

- Wipe C:\ProgramData\WPinternals\WPinternals.log
- Start WPinternals version 2.9.2
- Reproduce the issue
- Backup C:\ProgramData\WPinternals\WPinternals.log to C:\ProgramData\WPinternals\WPinternals_2.9.2.log
- Wipe C:\ProgramData\WPinternals\WPinternals.log
- Start WPinternals version 3.0 Prerelease 2
- Reproduce the issue
- Backup C:\ProgramData\WPinternals\WPinternals.log to C:\ProgramData\WPinternals\WPinternals_3.0.log
- Wipe C:\ProgramData\WPinternals\WPinternals.log

Send both WPinternals_3.0.log and WPinternals_2.9.2.log

Thank you.

EpicLPer, 2024-09-22:
Heya, I fixed the issue meanwhile and it seemed to be device related. So the only logs I'd have are the ones still being left behind after all my tests, tho I guess they would be huge to share here. If there's anything specific I can share tho or a specific section that would help can of course provide then :)

I had Windows 11 flashed on it before, somehow it was stuck in a somewhat "limbo state" of WPI telling me my bootloader is locked but it... wasn't? After using thor2 directly to flash a FFU for my device it flashed just fine, however was then stuck at a red flash recovery screen. WPI at that point picked it up again and suddenly the bootloader unlock steps started working :)
After like 20 failed attempts from WPI to try and unlock it it threw an error that it couldn't do so, however at that point Windows Mobile started booting... I then set everything up, shut down my 950 and then tried the bootloader unlock again which now worked.

So yeah, I think my Lumia was just in a somewhat weird semi-bricked state which confused WPI, would be my guess.

EpicLPer, 2024-09-22:
Here are some sections from the log that I think are the interesting bits, if you need more please tell :)

```
2024-09-22 03:07:55.034: Flash app: 2.75
2024-09-22 03:07:55.034: Flash protocol: 2.41
2024-09-22 03:07:55.034: SecureBoot: Disabled (Platform Secure Boot: Enabled, UEFI Secure Boot: Enabled)
2024-09-22 03:07:55.034: Flash app security: Enabled (FFU security: Enabled, RDC: Not found, Authenticated: False)
2024-09-22 03:07:55.034: JTAG: Disabled
2024-09-22 03:07:55.034: Bootloader: Lumia Bootloader Spec B
2024-09-22 03:07:55.034: ProductCode: 059X4X7
2024-09-22 03:07:55.034: ProductType: RM-1104
2024-09-22 03:07:59.997: FFU not added, because it was already present in the repository.
2024-09-22 03:07:59.997: Emergency files not added, because they were already present in the repository.
2024-09-22 03:07:59.997: Unlock Bootloader
2024-09-22 03:07:59.997: Processing resources:
2024-09-22 03:07:59.997: Profile FFU: C:\Users\epicl\Downloads\RM1104_1078.0053.10586.13169.13842.034EA5_retail_prod_signed.ffu
2024-09-22 03:07:59.997: EDE file: C:\Users\epicl\Downloads\MPRG8992_fh.ede
2024-09-22 03:08:00.160: Assembling data for unlock
2024-09-22 03:08:00.161: Assembling data for unlock
2024-09-22 03:08:00.189: Attempt patch: SecureBootHack-V2-EFIESP
2024-09-22 03:08:00.193: Pattern: 0, 0
2024-09-22 03:08:00.193: Pattern: 1, 0
2024-09-22 03:08:00.193: Pattern: 2, 0
2024-09-22 03:08:00.193: Pattern: 3, 0
2024-09-22 03:08:00.193: Pattern: 4, 0
2024-09-22 03:08:00.193: Apply: 10.0.10586.318
2024-09-22 03:08:00.197: Edit BCD
2024-09-22 03:08:00.267: Enabling Test Signing
2024-09-22 03:08:00.267: Enabling Test Signing
2024-09-22 03:08:00.274: Error: Object reference not set to an instance of an object.
2024-09-22 03:08:00.278:    at WPinternals.LumiaUnlockBootloaderViewModel.LumiaUnlockUEFI(PhoneNotifierViewModel Notifier, String ProfileFFUPath, String EDEPath, String SupportedFFUPath, SetWorkingStatus SetWorkingStatus, UpdateWorkingStatus UpdateWorkingStatus, ExitSuccess ExitSuccess, ExitFailure ExitFailure, Boolean ExperimentalSpecBEFIESPUnlock, Boolean ExperimentalSpecAEFIESPUnlock, Boolean ReUnlockDevice)
2024-09-22 03:08:00.278: Object reference not set to an instance of an object.
2024-09-22 03:08:10.678: Error: Object reference not set to an instance of an object.
2024-09-22 03:08:10.678:    at WPinternals.LumiaFlashRomViewModel.<>c__DisplayClass13_0.<FlashFFUTask>b__0()
   at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ThreadHelper.ThreadStart()
```

Sometime later in the log:

```
2024-09-22 03:10:11.058: Edit BCD
2024-09-22 03:10:11.125: Enabling Test Signing
2024-09-22 03:10:11.125: Enabling Test Signing
2024-09-22 03:10:11.132: Error: Object reference not set to an instance of an object.
2024-09-22 03:10:11.138:    at WPinternals.LumiaUnlockBootloaderViewModel.LumiaUnlockUEFI(PhoneNotifierViewModel Notifier, String ProfileFFUPath, String EDEPath, String SupportedFFUPath, SetWorkingStatus SetWorkingStatus, UpdateWorkingStatus UpdateWorkingStatus, ExitSuccess ExitSuccess, ExitFailure ExitFailure, Boolean ExperimentalSpecBEFIESPUnlock, Boolean ExperimentalSpecAEFIESPUnlock, Boolean ReUnlockDevice)
2024-09-22 03:10:11.138: Object reference not set to an instance of an object.
2024-09-22 03:10:24.889: Initializing flash...
2024-09-22 03:10:24.941: Error: Object reference not set to an instance of an object.
2024-09-22 03:10:24.943:    at WPinternals.LumiaV2UnlockBootViewModel.LumiaV2FlashPartitions(PhoneNotifierViewModel Notifier, String EFIESPPath, String MainOSPath, String DataPath, SetWorkingStatus SetWorkingStatus, UpdateWorkingStatus UpdateWorkingStatus, ExitSuccess ExitSuccess, ExitFailure ExitFailure)
2024-09-22 03:10:24.943: Object reference not set to an instance of an object.
2024-09-22 03:13:26.265: Effective Bootloader Security Status: False
2024-09-22 03:13:35.741: Effective Bootloader Security Status: False
```

Cheikhabbas44, 2024-09-22:
[image]

Cheikhabbas44, 2024-09-22:
[image]

Cheikhabbas44, 2024-09-22:
Bonjour, comment allez-vous ? J'ai un problème sur mon lumière 950. Je n'arrive pas à déverrouiller le chargeur du démarrage s'il vous plaît. Aidez-moi comment faire pour flasher mon téléphone merci vous voyez à chaque fois que j'ai essayé de flasher j'essaie qui s'affiche[image]

gus33000, 2024-09-22:
@EpicLPer these log sections could be useful but only if i knew what version of the app they matched for, could you try to fetch this? thank you.

gus33000, 2024-09-22:
@Cheikhabbas44 this is a separate issue you are having and is also offtopic and not in english, i would advise you to file a separate issue properly on this repo. thank you.

Cheikhabbas44, 2024-09-22:
Hello, how are you? I have a problem on my Lumia 950. I can't unlock the bootloader please. Help me how to flash my phone thank you you see every time I tried to flash I try that is displayed
[image]
[image]

gus33000, 2024-10-12:
@EpicLPer Please try to gather logs with preview 3 (https://github.com/ReneLergner/WPinternals/releases/tag/3.0-preview3)

Delete the old log file before, we added a lot more debug logging in this preview update. This should enable narrowing down the issue

EpicLPer, 2024-10-14:
I sadly cannot reproduce the issue anymore or test the new release as I've managed to flash my Lumia already, so it's essentially "back to stock" and then flashed with an Android ROM :( Whatever weird partition layout or other issues my Lumia had with WPInternals is therefore gone now.

gus33000, 2024-10-15:
Alright if you ever get the issue again please let us know via a new issue then, good to know it got solved on your side thank you!

closed 2024-10-15 by gus33000