ReneLergner/WPinternals #81: Lumia 430 DS bricks after failing to unlock bootloader

win98se, 2024-10-31:
### Description

As in title. Every unlock ends with an error message that reads “Error: Failed to write to pipe: 4E4F4B58464645”, and the phone is now bricked (can only enter the dark red color flash boot screen).

Tried with every port of two separate computers, also with different cables, even tried installing Windows Device Recovery Tool (WDRT) as some other guides suggested, all same.

Additional infos:
- The Lumia 430 DS I'm attempting to unlock is running Windows 10 Mobile Version 1709 (Build 15254.603).
- Original device FFU is downloaded from https://lumiafirmware.com/ser/RM-1099/guid/059X1D6 (I've confirmed that the hardware model and the product code should be correct, as given by WPinternals).
- Emergency programmer is downloaded from https://xdaforums.com/t/need-lumia-430-ede-and-edp-file.3689455/post-75085595.
- Donor FFU is downloaded from https://lumiafirmware.com/ser/RM-1085/guid/059X4T0.

### Steps To Reproduce

1. Select “Unlock bootloader” from the sidebar.
2. Choose RM1099_02074.00000.15234.28002_RETAIL_prod_signed_1008_0268C4_000-SG_MV.ffu as the original device FFU.
3. Choose MPRG8x12_fh.ede as the emergency programmer.
4. Choose RM1085_1078.0053.10586.13169.12742.034EE8_retail_prod_signed.ffu as the donor FFU.
5. Click on “Unlock”.

### Expected behavior

Unlocking bootloader is successful and the device should return to normal booting for next steps.

### WPinternals Version

2.9.2 & 3.0 preview 4

### Windows Version

Windows 11 Version 24H2 (Build 26100.2161)

### Phone model

Lumia 430 Dual SIM (RM-1099)

### Relevant Assets

Here are the two log files from two separate computers -
- [WPInternals.log](https://github.com/user-attachments/files/17580297/WPInternals.log)
- [WPInternals.log](https://github.com/user-attachments/files/17580299/WPInternals.log)

gus33000, 2024-11-03:
Hi, is your device plugged in into one of the following type of ports by any chance:

- Thunderbolt 2 capable USB port
- Thunderbolt 3 capable USB port
- Thunderbolt 4 capable USB port
- Thunderbolt 5 capable USB port
- USB-C Receptacle USB port
- USB 3.0 SuperSpeed USB port
- AMD Ryzen 1st & 2nd Gen XHCI Controller USB port
- VM passthrough USB port

Connecting a Lumia to any of the following above will cause severe issues regarding data transfers and speed due to some issues with the hardware implementation of the following controller and WPinternals tight reliance on reliable and fast data transfers during exploitation.

You need to either downgrade the connection on your host computer (for example, finding an USB 2.0 only port, or converting USB 3.0 to USB 2.0 by physically cutting the superspeed lines (or using a USB 2.0 HUB plugged into the USB 3.0 port) otherwise you are never going to get past this.

win98se, 2024-11-04:
> You need to use a reliable known good USB controller on the host

Are there any examples of such USB controllers? I may try finding such an environment for more tests.

gus33000, 2024-11-10:
Anything that isn't falling into the above bullet point list should be fine

win98se, 2024-11-17:
I finally got it back after using a third computer without any USB 3.x controllers (which has a 2nd gen processor, instead of the 12th and 4th gens). Thank you.

closed 2024-11-17 by win98se