ReneLergner/WPinternals #83: Bootloader unlock fails on Lumia 635

shiypc, 2024-12-18:
### Description

Attempting to unlock bootloader on a T-Mobile Lumia 635 (RM-975 059V6W5) results in the error "Sequence contains no matching element" and fails.

### Steps To Reproduce

1. Connect phone via USB
2. Start WPI
3. Click unlock bootloader and add necessary files (FFU, donor FFU, emergency programmer)
4. Wait and observe error

### Expected behavior

Phone expected to unlock normally

### WPinternals Version

3.0.9061.34992

### Windows Version

Windows 11 23H2 22621

### Phone model

Lumia 635 (RM-975 T-Mobile 512M variant)

### Relevant Assets

[WPInternals.log](https://github.com/user-attachments/files/18189202/WPInternals.log)

flbauer, 2025-11-30:
Hello shiypc,
I have tried to unlock bootloader of a Lumia 640 with different versions of WPInternals running on a W10-PC  with the same result. Message says that : "SecureBoot Unlock V2" needs to be installed. Did you meanwhile know how this can be done?
Thanks

shiypc, 2026-01-02:
> Hello shiypc, I have tried to unlock bootloader of a Lumia 640 with different versions of WPInternals running on a W10-PC with the same result. Message says that : "SecureBoot Unlock V2" needs to be installed. Did you meanwhile know how this can be done? Thanks

for me, reinstalling a different firmware worked as it was a model issue. you may have a different issue though as i did not get any errors related to "secureboot unlock v2"