ReneLergner/WPinternals #85: Can't open WPI

haigamingP, 2024-12-21:
[image]

jack-obrien, 2024-12-21:
Looks like this would explain it: "This release being a preview build comes with caveats, known issues, and will expire on December 21, 2024."

From the release notes back in 2021: https://github.com/ReneLergner/WPinternals/releases/tag/3.0-preview4

gus33000, 2024-12-21:
Prerelease versions are intended for evaluation purposes only and can have rough edges and be highly unstable to the point of breaking your device. This behavior is intended. If you care about your device please use the latest stable version; not preview. Otherwise; you can roll your date back at your own risk. This works as intended either way.

closed 2024-12-21 by gus33000