ReneLergner/WPinternals #86: No Firmware Image for Lumia 950 XL

SebRock124, 2024-12-22:
### Description

The Software is refusing to show files for the Lumia 950 XL Dual Sim

### Expected behavior

I wanted it to show the firmware image files.

### WPinternals Version

2.9

### Windows Version

Windows 11 Home Build 26100

### Phone model

Lumia 950 XL (RM-1116)

### Relevant Assets

[image]

gus33000, 2024-12-22:
Hi, Microsoft has retired completely the server that hosted and provided the rest api to download Lumia firmware files. As a result the application is not going to help you download files anymore. You can try your best to find a public mirror of said files for your phone (check out lumiafirmware.com or archive.org and try searching for your model number and product code).

There is nothing we can do here.

closed 2024-12-22 by gus33000

SebRock124, 2024-12-22:
Thank you. I can't believe they did that.