ReneLergner/WPinternals #89: Windows Phone Internals can't find RM-1105 FFU and recovery files

thanone2, 2025-01-29:
### Description

I tried using RM-1104 files, but it didn't work, and it's giving me: "Error 0x1304: Invalid platform ID"
and also, I forgot to check the phone after resetting it, but when I boot into W10M it doesn't show any thing other than the navigation bar, battery icon, and the no sim card icon, I can turn it off, it takes screen shots, but that's about it.

### Expected behavior

I expected it to not fail, but, idk what else to put here

### WPinternals Version

Latest version (date it was published: 12/23/2024)

### Windows Version

Windows 11 Pro 24H2 26100.2894

### Phone model

Lumia 950 AT&T

### Relevant Assets

[WPInternals.log](https://github.com/user-attachments/files/18581790/WPInternals.log)

thanone2, 2025-01-29:
oh, and windows device recovery app throws an error about proxy settings, I reinstalled it then it tells me to get the newest version of the app, which it is the newest version

thanone2, 2025-01-29:
there's only this, and it won't download it

[image]

thanone2, 2025-01-29:
I just did a hard factory reset, I'm gonna see what happens

thanone2, 2025-01-29:
ok, so resetting got me back to windows mobile working fine

closed 2025-01-29 by gus33000

thanone2, 2025-01-29:
wdym "not planned"?

gus33000, 2025-01-30:
Sorry there is only two choices on github; complete and "not planned" Given your issue is resolved and it wasnt exactly about this program, thats why it got closed

thanone2, 2025-01-30:
it's not resolved though, there is still no ffu for my lumia

thanone2, 2025-01-30:
I fixed the other problem I had, yeah, but the original problem still stands for me

thanone2, 2025-01-31:
I need the RM-1105 FFU for my phone and it's not giving me it... idk what to do @gus33000

gus33000, 2025-02-01:
Microsoft permanently shutdown the servers that hosted these files. That's why the program doesn't give anything to you. There's nothing we can do and it's not an issue with the program. You can search online for third party sites that would rehost those. (Like Lumia firmware). But there's no issue with this tool. This issue remains closed.

thanone2, 2025-02-01:
dang, ok, thanks