ReneLergner/WPinternals #101: Unlocking RM-892 bootloader gets stuck in "Switching to Emergency Download Mode..."

rr124B, 2025-08-17:
Im trying to unlock my Lumia 925 with WPI. The thing is, it doesnt want to. 

I select my ffu, emergency folder and sbl3 file. The tool boots phone into flashing mode and then say "Switching to Emergency Download Mode..." and keeps loading infinitely, rendering phone unuseable until i put it on red nokia screen with thor2 and flash the original .ffu with wpi (which ive done so much right now i think i can write that command on my own atp).

Ive tried on multiple computers, on windows 11, 10, 8.1. On AMD and intel devices and theres still no difference. Ive  deleted and reinstalled qualcomm drivers, wdrt and its drivers one by one. Still, no difference.

Mind you, i can use thor2 and flash original ffu without any problems. But i cant, for some reason, unlock this devices bootloader. 

Is there anything i can do or have i bought like the only un-unlock-able lumia 925 ever? Im desparate.

Thanks in advance.

gohoski, 2025-08-24:
@rr124B it seems that 2.9.x does not work properly on latest Windows 10/11, try running [3.0 (Preview 4) (Refresh)](https://github.com/ReneLergner/WPinternals/releases/tag/3.0-preview4.1) using [RunAsDate](https://www.nirsoft.net/utils/run_as_date.html) because it has a timebomb. 3.0 Preview 4 ran perfectly fine for me for my Lumia 630, while 2.9.2 was always getting stuck at assembling data/test signing/similar.

Basically open RunAsDate, select the wpiternals exe, select date from march/april 2025, click run as administrator then click run. The registration doesn't send data anywhere, just type in random info.

gus33000, 2025-08-25:
The log file would help a lot. Can you post it? You seem to have skipped the field for it when filing the issue form.

gus33000, 2025-08-25:
You also seem to have entirely erased said issue form now that I'm viewing this better too. Please refile and don't go outside the form as all info really helps

rr124B, 2025-09-07:
> The log file would help a lot. Can you post it? You seem to have skipped the field for it when filing the issue form.

Can you tell me where can i find it. I looked for it but couldn't find.

rr124B, 2025-09-07:
> You also seem to have entirely erased said issue form now that I'm viewing this better too. Please refile and don't go outside the form as all info really helps

Also idk what you mean. I only created github for this specific issue. Can you explain it like youre explaining it to a toddler? Im such a noob, sorry.

gus33000, 2025-09-09:
[image]

Select bug report when clicking on file new issue, the form will also guide you in order to get the log file

closed 2025-09-09 by gus33000